openpnp/openpnp #70: Job freezes after first component placed when no tapefeeder specified.

maaadbob, 2015-01-28:
First trial run setup on my machine consisted of movable head, vacuum source and a small set of 0805 parts from a cut tape.

First part placement consisted of the head moving to the part, lowering the nozzle to the correct height, activating the vacuum, lifting the part out of the tape, moving to the destination, lowering the part and switching the vacuum off.
At this point it the program was expected to lift the nozzle and move the head to the next part in the cut tape.
This did not happen.

I thought that maybe since tape feeders are expected to be used then perhaps one needs to be defined.

I added a reference tape feeder and added actuator A1 to it.  (The clue was in machine.xml which listed the actuator, but not a tapefeeder)

Now the machine picks and places all parts in the job BOM even though no tapefeeder nor actuator is present on the machine.

vonnieda, 2015-01-28:
Can you show your machine.xml before and after the change?

vonnieda, 2015-02-07:
@maaadbob Is this still happening or can I close this?

maaadbob, 2015-02-08:
Hi Jason,

I'll try to get my machine.xml file uploaded tomorrow.
Sorry for the delay. Am quite busy with work.

Br,
Bob.

Connected by MOTOBLUR™

https://github.com/openpnp/openpnp/issues/70#issuecomment-73385026

maaadbob, 2015-02-20:
*********machine.xml.bak*********
<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine">
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="H1">
            <nozzles>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="N1" pick-dwell-milliseconds="0" place-dwell-milliseconds="0">
                  <nozzletips/>
                  <head-offsets units="Millimeters" x="10.0" y="0.0" z="0.0" rotation="0.0"/>
                  <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="NT1" allow-incompatible-packages="true">
                     <compatible-package-ids class="java.util.ArrayList"/>
                  </nozzle-tip>
               </nozzle>
            </nozzles>
            <actuators>
               <actuator class="org.openpnp.machine.reference.ReferenceActuator" id="A1" index="0">
                  <head-offsets units="Millimeters" x="10.0" y="10.0" z="0.0" rotation="0.0"/>
               </actuator>
            </actuators>
            <cameras>
               <camera class="org.openpnp.machine.reference.camera.OpenCvCamera" id="C1" looking="Down" device-index="0">
                  <units-per-pixel units="Millimeters" x="0.032" y="0.032" z="0.0" rotation="90.0"/>
                  <vision-provider class="org.openpnp.machine.reference.vision.OpenCvVisionProvider"/>
                  <head-offsets units="Millimeters" x="20.0" y="10.0" z="0.0" rotation="0.0"/>
               </camera>
            </cameras>
         </head>
      </heads>
      <feeders>
         <feeder class="org.openpnp.machine.reference.feeder.ReferenceTapeFeeder" id="F1" enabled="false" part-id="C-0805-100NF">
            <location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            <feed-start-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            <feed-end-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            <feed-speed>1.0</feed-speed>
            <vision enabled="false">
               <area-of-interest x="0" y="0" width="0" height="0"/>
               <template-image-top-left units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
               <template-image-bottom-right units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            </vision>
         </feeder>
         <feeder class="org.openpnp.machine.reference.feeder.ReferenceTrayFeeder" id="F2" enabled="true" part-id="C0805-" tray-count-x="100" tray-count-y="1" feed-count="0">
            <location units="Millimeters" x="25.0" y="5.0" z="10.0" rotation="0.0"/>
            <offsets units="Millimeters" x="1.0" y="0.0" z="0.0" rotation="0.0"/>
         </feeder>
      </feeders>
      <cameras/>
      <job-planner class="org.openpnp.planner.SimpleJobPlanner"/>
      <job-processor class="org.openpnp.machine.reference.ReferenceJobProcessor"/>
      <driver class="org.openpnp.machine.reference.driver.GrblDriver" port-name="/dev/ttyACM0" baud="19200" feed-rate-mm-per-minute="2000.0"/>
   </machine>
</openpnp-machine>

*********machine.xml*********
<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine">
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="H1">
            <nozzles>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="N1" pick-dwell-milliseconds="1000" place-dwell-milliseconds="1000" current-nozzle-tip-id="NT1" changer-enabled="true">
                  <nozzle-tips>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="NT1" allow-incompatible-packages="true">
                        <compatible-package-ids class="java.util.HashSet">
                           <id>C0805</id>
                        </compatible-package-ids>
                        <changer-start-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                     </nozzle-tip>
                  </nozzle-tips>
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
[... 14 lines cut ...]
         </head>
      </heads>
      <feeders>
         <feeder class="org.openpnp.machine.reference.feeder.ReferenceTrayFeeder" id="F1" enabled="true" part-id="C0805-" tray-count-x="1" tray-count-y="10" feed-count="5">
            <location units="Millimeters" x="262.7" y="109.8" z="5.8" rotation="0.0"/>
            <offsets units="Millimeters" x="0.0" y="-4.0" z="0.0" rotation="0.0"/>
         </feeder>
         <feeder class="org.openpnp.machine.reference.feeder.ReferenceTapeFeeder" id="F2" enabled="true" part-id="C-0805-100NF" actuator-id="A1">
            <location units="Millimeters" x="212.0" y="50.0" z="10.0" rotation="0.0"/>
            <feed-start-location units="Millimeters" x="220.0" y="60.0" z="5.0" rotation="0.0"/>
            <feed-end-location units="Millimeters" x="220.0" y="64.0" z="5.0" rotation="0.0"/>
            <feed-speed>1.0</feed-speed>
            <vision enabled="false">
               <area-of-interest x="0" y="0" width="0" height="0"/>
               <template-image-top-left units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
               <template-image-bottom-right units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
            </vision>
         </feeder>
      </feeders>
      <cameras/>
      <job-planner class="org.openpnp.planner.SimpleJobPlanner"/>
      <job-processor class="org.openpnp.machine.reference.ReferenceJobProcessor"/>
      <driver class="org.openpnp.machine.reference.driver.GrblDriver" port-name="/dev/ttyACM0" baud="19200" feed-rate-mm-per-minute="1000.0"/>
   </machine>
</openpnp-machine>

vonnieda, 2016-02-13:
Note self: Try to duplicate using the posted machine.xml.

vonnieda, 2016-02-15:
This is actually (mostly) a configuration problem. In the first machine.xml there is an error in defining the nozzle tip, which caused no nozzle tip to be available for a placement, causing a feed related error which the system used to silently ignore. This has been fixed by several previous commits.

closed 2016-02-15 by vonnieda