openpnp/openpnp #73: Camera enumeration issues (OpenCV upstream)

FlyingLotus1983, 2015-02-03:
This OpenCV bug has been open for almost 4 years:
http://code.opencv.org/issues/935

Currently right now, there's no way to ensure that your cameras are going to come up in the correct order unless you unplug them all, and plug them in, in the correct order.  

I'm trying to get the opencv guys to escalate the issue, and am even trying to put a cash bounty on it.  Figured putting this issue here would help us keep track of it.

cri-s, 2015-02-03:
On Windows and Unix solution exists for usb devices.
What always works and it is Platformindipendent is fingerprinting the webcam.
So if different models are used, this are recognized.
What Os do you use, so i make example of fingerprinting and enumeration.

FlyingLotus1983, 2015-02-03:
Fingerprinting sounds interesting.  I'm on windows for my dev machine, but
will be using Linux and Mac eventually too for OpenPNP.

cri-s, 2015-02-04:
After thinking about it i came to conclusion that it is not needed.
The opencv gives back how many cameras are installed. There can be sequencially accessed from 
1 to max number of camera.
If having downlooking camera only the cam can be identified using homing dot.
Instead of dot, template matching should be used, and all cameras should be checked for best match.
If having both down and uplooking camera the homing dot could be both cameras as example.
If having uplooking camera only the nozzle or somoe part of nozzle holder should used for finding cam.
Further this increases machine reliability.
Note: there should be added function to set Axis Offset, emitting  G92 on the driver side.
Note: there should be bool variable for each nozzle if pos was calibrated in order to handle multidriver/head setups and for loop like NozzleZero on machine. Only Homing should clear flag.

vonnieda, 2015-02-04:
I'm not aware of OpenCV indicating how many cameras are installed. Do you
have a reference for that?

cri-s, 2015-02-04:
you could use setDevice and capture to test if camera is present. 
Most cameras return on the first call null, do capture twice.
The only piece missing is function to close videocapture, functionality that should exist and called
if closing the program. Otherwise it is possible that the camera cannot be accessed the next time 
until rebootet or unplugged.

/**
 * Get the number of camera available
 */
int countCameras()
{
   cv::VideoCapture temp_camera;
   int maxTested = 10;
   for (int i = 0; i < maxTested; i++){
     cv::VideoCapture temp_camera(i);
     bool res = (!temp_camera.isOpened());
     temp_camera.release();
     if (res)
     {
       return i;
     }
   }
   return maxTested;
}

cri-s, 2015-02-04:
Adapted to java:

        private
        int countCameras()
        {
                        for (int i = 0; i < 10; i++){
                        try{
                                fg.open(i);
                                if(fg.isOpened()) fg.release();
                                else return i;
                        }       catch (Exception e) {
                                        return i;
                                }
                        }
                return 10;
        }

cri-s, 2015-02-04:
yes it works, but gives some problem, i must check it. just wait

FlyingLotus1983, 2015-02-04:
There are some people that have tried this approach and it does not work
well in all cases ( Mac / Windows / Linux).  If I remember where I saw the
link, I will post it here.

cri-s, 2015-02-04:
This approach works for enumerated cams without problems on any platform.
I have seen some code where there don't count from 0 but from -1 .
That don't work
because -1 is special and don't mean always cam 0.
It don't work for non enumerated cams, Ethernetcams, Wifi Cam,
Streaming, Videofiles.

FlyingLotus1983, 2015-02-04:
OK, so assuming the code to enumerate the # of cameras works... What about
distinguishing one camera from the other?  I have two cams, both 720p, but
different brands, different USB ID's etc.  If OpenCV were to add info
params for each camera, I could check and always tie a particular USB ID to
a camera identifier "C1", "C2", etc.  I'm not aware of any workarounds
other than just fixing/enchancing the OpenCV code.  Your approach to
identify based off of visual data on the machine sounds great, but will
take a lot of time and debugging,and even then won't always work.

vonnieda, 2016-01-13:
New LibuvcCamera solves this issue, and it's not likely OpenCV is ever going to improve this API, so I am closing this ticket.

closed 2016-01-13 by vonnieda