openpnp/openpnp #77: Out of memory error when using mulitple cameras

FlyingLotus1983, 2015-02-04:
I'm getting an out of memory error for some reason, thinking it might have to do with multiple cameras.  I initially had a 90 degree rotation on C1, but the memory error happens if set to zero (which I assume bypasses the rotation code?).

When disabling one camera or the other camera, the system seems to work as expected.
With both cameras enabled, it will eventually freeze with the error below.. the time-to-freeze varies, but is never more than a few minutes.

```
Exception in thread "Thread-4" java.lang.OutOfMemoryError: Java heap space
at java.awt.image.DataBufferByte.<init>(DataBufferByte.java:76)
at java.awt.image.Raster.createInterleavedRaster(Raster.java:266)
at java.awt.image.BufferedImage.<init>(BufferedImage.java:384)
at org.openpnp.util.OpenCvUtils.toBufferedImage(OpenCvUtils.java:43)
at org.openpnp.machine.reference.camera.OpenCvCamera.capture(OpenCvCamera.java:68)
at org.openpnp.machine.reference.camera.OpenCvCamera.run(OpenCvCamera.java:86)
at java.lang.Thread.run(Thread.java:745)
```

My machine.xml is posted here:
https://gist.github.com/FlyingLotus1983/51747c6de78558836a9d

FlyingLotus1983, 2015-02-04:
I've never troubleshooted a Java memory error, so I'm not really sure where to start.  So far the Oracle docs aren't that much of a help.

FlyingLotus1983, 2015-02-04:
At Jason's request, I tried adding -Xmx1024m  to the JVM options, which increases the available memory to the process.  

Still dies with 2 cameras. 
Never, ever dies with 1 camera.
Still investigating and checking code for potential problems that could cause this.

Oh, this is a Windows issue BTW :)

cri-s, 2015-02-04:
Inside OpenCv capture, there is always create new Mat and never released it.
Or release it, or use always the same, or if using double buffering, the same two Mat.
mat.release() should resolve the issue.  I don't have investigated how doing it the correct way.
possible solution:

    private Mat capMat;
        @Override^M
        public synchronized BufferedImage capture() {^M
            if (thread == null) {^M
                setDeviceIndex(deviceIndex);^M
            }^M
                try {^M
                    if(capMat!=null) capMat.release();
                    capMat = new Mat();^M
                    if (!fg.read(capMat)) {^M
                        return null;^M
                    }            ^M
                        return OpenCvUtils.toBufferedImage(capMat);^M
                }^M
                catch (Exception e) {^M
                        return null;^M
                }^M
        }^M
        ^M

The ^M are just dos/unix artifact, remove it.

cri-s, 2015-02-04:
Give feedback if this resolves the problem.

FlyingLotus1983, 2015-02-04:
I will give this a try tonight, thanks!!

vonnieda, 2015-02-04:
Good catch. I didn't realize they had to be manually released. If that's
the case then this is definitely a memory leak. I'm surprised the library
doesn't handle that with a finalizer though.

vonnieda, 2015-02-04:
If this works, it can actually be simplified. Just release the Mat after
calling OpenCvUtils.toBufferedImage(capMat).
OpenCvUtils.toBufferedImage(capMat)
copies the data out of the Mat, so there is no need to have it afterwards.

cri-s, 2015-02-04:
The capture theretically should return rectified Mat.

vonnieda, 2015-02-04:
> The capture theretically should return rectified Mat.

Can you explain further? I don't understand what you mean by this.

vonnieda, 2015-06-30:
@FlyingLotus1983 Is this issue still valid? I see you had committed a fix. Are you using it, or are you using the stock code?

FlyingLotus1983, 2015-06-30:
I'll try to check and get back to you.. Don't think I ever submitted a pull request for it.  Will investigate.

FlyingLotus1983, 2015-07-20:
OK, just looked into this, I had committed it to my old fork but never did a pull request.  I just applied the diff and will send a pull request for this, just as soon as I finish getting eclipse running on my new machine.

vonnieda, 2015-10-04:
This has been fixed for quite a while. The Mat is now released before the function returns.

closed 2015-10-04 by vonnieda