openpnp/openpnp #78: Save job.xml and board.xml every 'x' minutes, not just on close

FlyingLotus1983, 2015-02-04:
If OpenPnP crashes, any work will be lost.  A better approach is to have OpenPnP save the job.xml and board.xml periodically, instead of just on exit.

cri-s, 2015-02-04:
Please don't save to same name. It is common to open job/board and to modify it without wanting to save.
Further on linux it may be possible that the file is read only. Actually openpnp don't start if there is no
internet connection when the default tablescanner is configured.
Saving actual job and board files on ~/.openpnp can be interesting, and if openpnp crashes user can
load it from there.

vonnieda, 2015-02-04:
Neil, when we first discussed this I thought you were referring just to the
OpenPnP databases (parts.xml, packages.xml, machine.xml) and not the job
and board. Personally I don't think we should try to save the job and
board. In my mind it's no different than needing to remember to save your
Word docs or whatever. It bugs me when programs try to auto-save and
clutter up my filesystem with backup files and autosave files.

vonnieda, 2015-10-04:
Closing this as won't fix. I don't want to randomly overwrite users work without asking them. If someone wants to submit a pull request that adds backup functionality I'll be happy to accept it, but I don't think this is necessary.

closed 2015-10-04 by vonnieda