openpnp/openpnp #80: Why need Camera have reference to VisionProvider

cri-s, 2015-02-06:
For me, it is not logical that camera are bound to visionprovider.
Is there some reason behind ?
Example if vision have  Location locateTemplate(Mat template, Mat image, byte type)

and the code calls
  Location offsets = vision.locateTemplate(part.getTemplate(),camera.picture(roi),part.getVisionType());
  offsets = camera.vision(offsets); // vision could be calcOffset, adjOffset, calcPos, adjPos, ...

eventually Vision could be attached to feeder, but not camera.
Sure camera could do coordinate translation , but it can be just one function to call.
It is more, that if internal function apply, then use it, otherwise use the next vision provider.
Vision provider can be SURF, .... if Law permit it, or other not so fast solution finder.
Eventually vision provider chain the calling queue itself.

vonnieda, 2015-02-06:
When I designed this API I didn't really have a clear idea of what I wanted
it to. It's been a temporary API that has lasted too long. I'm intending to
rework it quite a bit after the release of OpenPnP 1.0. It will become more
generic - similar to what you mentioned above. There will also be a
VisionManager that handles the work of collecting images from the
appropriate cameras and feeding them to the vision provider. That will be
the high level interface and VisionProvider will be the low level interface.

vonnieda, 2015-02-06:
Closing this as it's better suited as a discussion on the mailing list instead of in the issue tracker.

closed 2015-02-06 by vonnieda