openpnp/openpnp #81: OpenCV Camera, id should be Text field

cri-s, 2015-02-08:
As the Title Say, the ID should be text field.
It should be possible to enter -1, or as example http://178.169.93.71//axis-cgi/mjpg/video.cgi

I have trayed to replace ComboBox with TextField, but after starting the program the next time,
it have the value 0, maybe some trigger or validator is active.

FlyingLotus1983, 2015-02-08:
Karl created a camera class called org.firepick.FireFuseCamera that used a URL similar to your example above.  Rather than putting the URl in the camera ID, he added as a child element to the camera driver element.  For example:

```xml
            <cameras>
               <camera class="org.firepick.FireFuseCamera" id="C1" looking="Down" fps="1">
                  <units-per-pixel units="Millimeters" x="0.030745" y="0.030813" z="0.0" rotation="0.0"/>
                  <vision-provider class="org.firepick.FireSightVisionProvider"/>
                  <head-offsets units="Millimeters" x="0.0" y="-28.941" z="0.0" rotation="0.0"/>
                  <source-uri>http://10.0.1.4:8080/firerest/sync/cv/1/camera.jpg</source-uri>
               </camera>
            </cameras>
```

This was an older machine.xml so it might not be 100% usable but you get the idea. The `source-uri` holds the URL... Not the camera ID.

cri-s, 2015-02-08:
Do you sugget to Autoproble from -1 to 2000 and every camera detected is on the DeviceID ?

vonnieda, 2015-02-08:
This wouldn't work for the primary use case, which is using a local video
capture device. See:
http://docs.opencv.org/modules/highgui/doc/reading_and_writing_images_and_video.html#videocapture-open

OpenPnP calls bool VideoCapture::open(int device).

If you need to call bool VideoCapture::open(const string& filename) we
could add a second field called filename.

Jason

cri-s, 2015-02-08:
It is not less difficult to have one text field and doing something like that:
Integer id ;
String ids[] = index.split(" ");
try { id=Integer.valueOf(ids[0]);} catch(Exception e) { id=null; }
and then
if (id==null) fg.open(Index); else fg.open(id);

Further the textfield can be hook for adding option, like resolution, disable automatic gain, ... 
Default for the text field should be -1

vonnieda, 2015-02-08:
Sounds fine to me.

cri-s, 2015-07-21:
clenup

closed 2015-07-21 by cri-s