openpnp/openpnp #82: Jog controls (and other operator controls) should be disabled when a job is running.

vonnieda, 2015-02-08:
We need to have a standard interface that all operator initiated movements go through so that it can be disabled in a single place. This will allow us to disable that interface when a job is running so that the user can't accidentally move the machine mid-job.

vonnieda, 2015-02-09:
Come to think of it, all movement, including job movement, should go through the interface. Then whoever needs to work on the machine can just attempt to lock it for the length of the operation.

vonnieda, 2015-02-21:
Consider just using syncronization to do this, just lock the machine object and make sure all driver methods obtain the lock.

vonnieda, 2015-06-24:
Relates to #37 and #36

vonnieda, 2015-10-06:
Probably need a way to build a tree of ActionGroup to get this working right. There could be root level groups such as machineEnabled and machineBusy that other groups could extend. When a node becomes enabled or disabled all of it's children do as well.

mlyle, 2015-10-06:
> Consider just using syncronization to do this, just lock the machine object and make sure all driver methods obtain the lock.

This seems problematic because then the jog command will "hang" trying to obtain a blocking lock, won't it?  And then the jog will happen / effectively be queued after completion.

vonnieda, 2015-10-07:
@mlyle Agreed, I will use the new Machine.submit API along with a Machine busy check.

cri-s, 2015-10-07:
I have included MachineControlsPanel setEnabled into updateJobState
and included JogControlsPanel.setEnabled inside MachineControlsPanel
setEnabled .
Further updated actions inside JobPanel (camera/tool move)
and update code inside gui/components/CameraView.java for mouse click events.

I'm actually tweaking some code before making pull request.
couple of days.

vonnieda, 2015-10-07:
@cri-s Sounds good, looking forward to it!

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda