openpnp/openpnp #83: Add a diagnostic tool for helping people submit bug reports with the required data.

vonnieda, 2015-02-12:
When submitting a bug report, or even asking for help, it is always helpful to be able to look at the user's machine.xml, log files and in some cases their job and board files. We should give the user the ability to submit a diagnostic package which will include all of this information to make it easier to gather.

vonnieda, 2015-02-12:
This should also include some basic system information (Java version, OS version) and build information about OpenPnP. A Git hash would be ideal.

FlyingLotus1983, 2015-02-12:
Some other things, would be nice to have the option to give the option to
send parts.xml, packages.xml, board/job.xml, and maybe let the user choose
what to send, depending on the type of problem.

Also, if an array of mail recipients was specified in the Java code
somewhere, I'd be able to add a FirePick Delta bug mailbox from within our
fork.  So people using our fork could send bugs to us, and other forks
could setup where their reports go to.

One other thing i just thought of.  If we used the Github API, what about
the ability to log bugs right from OpenPNP?  Their API is pretty simple.

vonnieda, 2015-02-12:
I would lean towards using the GitHub API over email entirely. Email is surprisingly tricky to do from desktop programs unless you just launch the user's email editor and I don't want to deal with that. Gists might work, since they can be created anonymously. Otherwise the user will need to provide Github credentials or authorize OpenPnP using OAuth.

FlyingLotus1983, 2015-02-12:
Looking at https://developer.github.com/v3/issues/#create-an-issue , it seems relatively easy to post an issue.  I don't think it would be required to edit issues from openpnp, once people are subscribed, they should get email issues, and the repo maintainers can do all the tagging, milestones, closing, etc.

There are existing github libraries for Java: https://developer.github.com/libraries/

The GitHub Java API (org.eclipse.egit.github.core), is available in Maven.  Also there's the JCabi GitHub API is based on Java7 JSON API (JSR-353), and apparently covers the entire API.

FlyingLotus1983, 2015-02-12:
Gists are part of the API too, it would make sense to probably throw the machine.xml and such up in a gist, and then link that to the issue.

vonnieda, 2015-02-12:
The difference is that Gists can be created anonymously but issues require
authentication. So the user would have to have a Github account (not always
the case) and would have to authenticate OpenPnP to it before submitting
the issue. That might be too much to put in the way.

FlyingLotus1983, 2015-02-12:
I don't think most users would object to putting in their github username and password, as it's pretty safe to do, is secure, and is revokable via the github panel on their website.  If the github Java library were used, it's not difficult to do the authorizing, it's just a quick plug and chug of a few function calls and whatnot.

vonnieda, 2016-02-18:
Some ideas on implementing this:

1. Add a "Begin Bug Report" menu option. It will pop a dialog explaining the process, turn on debug logging, set some kind of checkpoint in the log, maybe create a new directory to store images in, and then tell the user to perform the actions that cause the bug.
2. User runs through the process that causes the bug.
3. User now clicks the "Finish Bug Report" button which pops a dialog showing what data will be collected with checkboxes. We want all three config files, the latest log file and any debug images that were captured.
4. The tool packs all of this data into a zipfile, uploads it to S3 and then creates a Github issue including a link to the package of data.

vonnieda, 2017-03-22:
This is now mostly done in https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2017-03-21. A few bits still unfinished are marked in the main comment.

vonnieda, 2017-03-22:
Fully done now.

vonnieda, 2017-03-22:
Bug in sorting when no images are present: https://gist.github.com/la3pna/776c1a0a8dd4795452bf0cf70d365302

LA3PNA also reports getting this dialog: http://imgur.com/a/dCxJZ. I feel like that could possibly be associated to #442 since I can't think of why the Submit Diagnostics code would cause a VM error like that.

vonnieda, 2017-04-02:
All fixed.

closed 2017-04-02 by vonnieda