openpnp/openpnp #84: Windows openpnp.bat some times fails to initiate the correct dll

la3pna, 2015-02-19:
The openpnp.bat file some times fails to initate the correct dll with the current use of 
This issue should only be a problem on Intel based machines. 
The %PROCESSOR_ARCHITECTURE% returns AMD64 on all 64 bit systems. Suggest to move to using %PROCESSOR_IDENTIFIER% instead. 
For AMD64 this returns: 
AMD64 Family 22 Model 48 Stepping 1, AuthenicAMD
For Intel64 this returns:
Intel64 Family 6 Model 58 Stepping 9, GenuineIntel

The complete bash file would then be: 

@echo off
set _test=%PROCESSOR_IDENTIFIER%
set archi=%_test:~0,3%

if not x%archi:X86=%==x%archi% java -Djava.library.path=lib\native\windows\x86 -jar target\openpnp-gui-0.0.1-alpha-SNAPSHOT.jar
if not x%archi:Int=%==x%archi% java -Djava.library.path=lib\native\windows\x86 -jar target\openpnp-gui-0.0.1-alpha-SNAPSHOT.jar
if not x%archi:AMD=%==x%archi% java -Djava.library.path=lib\native\windows\x64 -jar target\openpnp-gui-0.0.1-alpha-SNAPSHOT.jar

la3pna, 2015-02-22:
This gives the 64 bit on my win7 64, but this machine (Intel i5 64 bit) needs the 32 bit driver to work.
Not sure if this is a isolated problem or if its related to Intel?

vonnieda, 2015-02-22:
Oh, that's interesting. Are you running 32 bit Windows on a 64 bit
processor, by chance? Can you look in your system properties and check?

Thanks,
Jason

la3pna, 2015-02-22:
Just had to look, and I'm running 64 bit Win 7.

vonnieda, 2015-02-22:
Ah ha! So you must be running a 32 bit JVM?

vonnieda, 2015-02-22:
Following up from IRC conversation, was indeed 32 bit JVM and switching to a 64 bit VM fixed it.

closed 2015-02-22 by vonnieda