openpnp/openpnp #87: OpenCVCamera crash in Linux

vonnieda, 2015-04-21:
From email thread:

If you're bored and want to look at this, I quickly researched the camera error and everything seems to point to OpenCV code usage. An example...http://answers.opencv.org/question/25761/camera-is-not-getting-release/

I get this at the beginning...
HIGHGUI ERROR: V4L/V4L2: VIDIOC_S_CROP
HIGHGUI ERROR: V4L2: Pixel format of incoming image is unsupported by OpenCV

Then hundreds of these...
VIDIOC_STREAMON: Bad file descriptor

And an intermittent this... 
VIDIOC_STREAMON: Inappropriate ioctl for device

After a minute or two, the video freezes.  I'm using an endoscope camera and dmesg shows me this...
usb 1-8: new high-speed USB device number 6 using ehci_hcd
[21005.816575] usb 1-8: New USB device found, idVendor=090c, idProduct=f37d
[21005.816585] usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[21005.816592] usb 1-8: Product: HD Endoscope Camera
[21005.816598] usb 1-8: Manufacturer: Endoscope
[21005.817792] uvcvideo: Found UVC 1.00 device HD Endoscope Camera (090c:f37d)
[21005.854225] input: HD Endoscope Camera as /devices/pci0000:00/0000:00:1d.7/usb1/1-8/1-8:1.0/input/input9

Not sure if there's anything I can do at this point.  Is this something in code or otherwise in your control, or should I be looking elsewhere?

cri-s, 2015-04-22:
can you report output result from
guvcview -v -o
after that, retray running with adding this to env, depending on shell
it needs export before.
 LD_PRELOAD=/usr/lib/libv4l/v4l1compat.so

vonnieda, 2015-12-29:
No further reports of this since upgrading OpenCV. Closing until someone else brings it up.

vonnieda, 2017-02-20:
This appears to still be an issue in certain Linux configurations. More discussion in https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/t4THPd7n1sI/iR840uHGCwAJ. Of note, it seems that the problem lies in opening the camera after it's been opened. It seems like we're not shutting it down well enough before opening again.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda