openpnp/openpnp #100: Automatic Strip Feeder Configuration

vonnieda, 2015-06-23:
Use vision to automatically locate, orient and define a cut strip feeder.

vonnieda, 2015-06-23:
Early notes:

Strip Setup
---
* All it should take is clicking on a hole.
* We can try a black/white threshold and then a white/black threshold to figure out if it's white or black tape.
* Then search nearby for the next pitch distance hole to determine orientation. Then binary search to find end of tape.
* Then we need to try to find the part, which might require some input from the user, but think about it more.

la3pna, 2015-06-24:
How will this work on clear plastic tape like some of the larger passives
use?

vonnieda, 2015-06-25:
@la3pna Hadn't considered that. I suspect it will still be possible to find the holes even though the contrast will be quite limited. Just have to try.

FlyingLotus1983, 2015-06-25:
Short-term it wouldn't work as well as white paper tape.  But long-term it should be a piece of cake.  Might require some machine learning and other CV techniques, but it can be done.  I say get it working with white paper tape first, which will help prove out the RANSAC or whatever other algorithms calculate the direction and distance.  Then we can go back later and make it work with other tape materials.

vonnieda, 2015-06-26:
@la3pna @FlyingLotus1983 I ran across this post last night which I think answers the question pretty well: http://www.liteplacer.com/phpBB/viewtopic.php?f=11&t=7&start=10#p81

vonnieda, 2015-11-21:
Calling this done for now. The current implementation works pretty well.

closed 2015-11-21 by vonnieda