openpnp/openpnp #102: Pick Fail Detect and Discard

vonnieda, 2015-06-23:
We need the ability to detect a failed pick via various methods and then the ability to discard to a pit.

* The discard position should be configurable.
* Whether or not to discard should be an option, and possibly an option per feeder. Perhaps the options include Stop Job, Pause Job, Discard, Alert Operator, etc.
* This will require changes to the Driver interface, JobProcessor, MachineListener and probably others.

vonnieda, 2016-02-02:
Another useful comment from IRC is to detect both pick and place failure:
* On pick, check if we picked and if not retry
* Before place make sure we still have the part and haven't lost it along the way
* After place make sure the part is now off the tip and if it's not either discard or try to re-place

rene-dev, 2016-02-04:
Agree on the waste location in machine options. My machine has 2 locations configurable: one for actual waste, and one for components which are marked expensive, and are manually placed back in the tray/tube.
also, in the feeders menu, there is a pick button. there should also be a place/waste button. Might be helpful while setting up feeders.

vonnieda, 2016-05-14:
Fixed in #265. If a pick fails we now discard and replan.

vonnieda, 2016-08-08:
This is being reopened so that we can discuss vacuum pressure sensing with regards to pick fail detection.

vonnieda, 2016-11-17:
Vacuum detection is now implemented and allows for pick failure and place failure detection as of #373 . Still need the options discussed in the first comment.

vonnieda, 2019-05-27:
Fixed in https://github.com/openpnp/openpnp/commit/97e5bfc54827965a24eb710fa0932c3cf46ce29f

commit 35aa639a (2019-05-27): * Added missing signaler signals for job running and stopped.
* IMPORTANT CHANGE: The entire job processor has been refactored to allow for finer granularity in steps, better error handling including the ability to be context sensitive, better retry options for pick and feed failures, improved vacuum checking, and increased flexibility:
    * The FSM implemention that caused so many difficult to debug errors has been removed.
    * Each step can now be it's own class, and can maintain it's own state. This allows each step to determine it's own level of granularity.
    * Steps now direct the j [... cut]
closed 2019-05-27 by vonnieda