openpnp/openpnp #95: OpenBuilds Driver Miscalculating Position

vonnieda, 2015-06-23:


vonnieda, 2015-06-23:
Code comment that triggered this issue:

        // TODO: Somewhere in this code the offsets are not being applied. When
        // we try to move Z to -11.5 (exactly) and we're at 0 nothing happens
        // because this check fails. Probably need ot make sure we're considering
        // offsets, maybe by using getLocation() instead of using the coordinates
        // directly.

cri-s, 2016-01-24:
This is because of Z height and Cam angle are compared, and if both have same value, as it is the case
of  -11.5, then no movement is made.
 if (!Double.isNaN(z) && z != this.z) { // z is in mm
.. calc ..
        this.z = a;  // z is in degree
}

commit 501fe078 (2016-02-13): OpenBuilds driver now calculates it's Z position for comparison in degrees instead of... not at all. Fixes #95.
Fixes a bug where when using spring loaded homing the Z would not reset correctly when it is extended.
Fixes a homing timeout bug.
closed 2016-02-13 by vonnieda