openpnp/openpnp #96: Improve Safe Z

vonnieda, 2015-06-23:
Currently Safe Z is just fixed at 0. This was put in temporarily and it stuck.

Safe Z should instead be based on the height of all the movements in the job, combined with the height of any currently picked part.

Ideally, the Safe Z would start low and then move up as taller parts were picked for a job.

vonnieda, 2015-06-23:
Code comment that triggered this issue:

	/**
	 * TODO: Instead of constantly moving to Safe Z, instead, start with Safe
	 * Z and then lower it based on the height of the tallest part placed so far.
	 * Also need to take into account feeder height and fixtures.
	 * 
	 * This all being said, other people have said that they used a fixed Safe Z.
	 * So maybe this is not an important optimization.
	 */

vonnieda, 2016-01-19:
Neil's changes to add Safe Z to all the HeadMountables were merged in. Leaving this open as the original idea was to make this somewhat dynamic.

vonnieda, 2017-03-22:
Happy with the current system, no need for additional changes.

closed 2017-03-22 by vonnieda

irenji, 2018-02-28:
Hi , i ,m setting z safe and nozzle in picking go to top of feeder , then move to 0 height then pick a part and next go to z safe again , why nozzle go to 0 height befor picking ?

vonnieda, 2018-02-28:
@irenji If you are using vision, it's probably this: https://github.com/openpnp/openpnp/wiki/GcodeDriver#nozzle-is-moving-or-turning-when-camera-is-moved