openpnp/openpnp #97: JobPlanner Should Take Part Height Into Account

vonnieda, 2015-06-23:
This is just an idea that needs exploring:

     * 
     * TODO: Plan paths to place shorter parts before taller parts so that we
     * don't bump the tall ones. This will help optimize the changes to
     * the processor allowing a lower Safe Z.

vonnieda, 2015-06-23:
This is roughly related to #96

FlyingLotus1983, 2015-06-23:
Would this just be a matter of re-ordering the placements shown to the user?  Like an 'optimize' button, reminiscent of the Excel A->Z sort button (or it just happens on import)?  The idea there being that the user would ultimately have control over the order but the button would do a pretty good job of ordering them.

This would seem more logical than having the planner figure this out on the fly, which might lead to semi-unpredictable results.  However, I have no experience with what real PnP machines do.. take all this with a grain of salt, just my $0.02.

vonnieda, 2015-06-23:
@FlyingLotus1983 that makes a lot of sense, and may actually be a good way to "influence" the planner. I say influence because ultimately the planner has to make some decisions about order due to optimizing nozzle changes. But, the order of the placements would be a good way to add another weight to the planner's decisions.

And your point as to user control is probably important. I imagined the planner would be the  authority, but I bet users will want to override it.

So, with that, maybe it makes sense to have the "optimize" button run the planner to get an optimal setup and then let the actual order come from the table.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome. This is going to be handled by letting the user specify order in #280

closed 2017-04-13 by vonnieda