openpnp/openpnp #98: Add Z Touch Off

vonnieda, 2015-06-23:
This is several new features that should probably be broken out into their own issues.

1. The ability to detect a Z probe strike and create an event.
2. UI to show when a Z probe is triggered.
3. Automatic Z probe touch off.

vonnieda, 2015-06-23:
Initial thoughts regarding this feature:

	
	/**
	 * TODO: Add a way to z-probe. Ideally this would be automatic but we
	 * probably also want to be able to give UI feedback. So, perhaps we
	 * add:
	 * boolean isTouched()
	 * 
	 * something to the machine listener to fire an event when touch happens
	 * On that note, we should get rid of MachineListener and add events
	 * directly to the primary objects. Or we could add more discrete events
	 * to MachineListener such as nozzleTouched(Nozzle) and such.
	 * 
	 * void touchOff() which will run a zprobe and return when it's finished. It
	 * should adjust the nozzle's location to account for any required offset.
	 * In other words, if the zprobe requires compressing a spring, that
	 * compression offset should be subtracted from the location before it
	 * returns so that the final location is where the nozzle is just touching
	 * the probed location. This may require per-nozzletip configuration, so
	 * need to think about that too.
	 */

vonnieda, 2015-07-30:
New thinking on this is to make a new type of HeadMountable called Probe or Zprobe that would have it's own API, a la Nozzle, Camera or Actuator. This would allow the probe to be selected, deployed, etc. Likely any move against the probe would deploy it first.

vonnieda, 2016-02-02:
Current thinking, based on some chatter in IRC is to do this with pressure first. The problem of accidentally sucking up parts can be solved by touching off against the tape instead of the part, since it should be the same height. We can also always use the smallest nozzle tip for z-probe to reduce the risk.

vonnieda, 2016-05-26:
We'll probably need object specific z-probe systems. For instance, when setting the height of a board you probably don't want to probe the board corner because the board may be rounded - there might not be anything there. So instead it would make sense to probe the nearest placement, or something like that.

For feeders we'll want to probe the tape instead of the part so that we don't suck up a part.

We also need the ability to measure part height automatically, and that might include a pick, probe against a defined location and then discard. Or perhaps we just probe parts during their first usage in a job. Probably better to do the former, since we need height for other things like setting up bottom vision.

vonnieda, 2016-10-25:
Good ideas from this video https://www.youtube.com/watch?v=3kkf6bktIHA&feature=youtu.be&t=37s

Specifically, it looks like they have started the job without knowing the feeder, board or part height. It probes to the part to do the first pick, then probes to a metal block to determine the part height and then probes to the board to determine the board height. This happens in the first pick for each feeder and then the machine moves to full speed placement.

We could come up with a value or just a flag for each of those values that indicates the job processor should probe rather than throwing an error for 0 values, assuming a probe is attached.

issus, 2016-10-25:
Thats pretty much how Liteplacer does it.

Mike-Menci, 2018-11-20:
Here a test : https://www.youtube.com/watch?v=-1168zeOEH8
and one more demonstration with Juki: https://www.youtube.com/watch?v=DO8lbh_2isU
Coding - ? Who can do it ?

markmaker, 2021-05-03:
Most of this has been done through #1174.

But the touch plate support remains to be done. I see two implementations:

- A touch plate with built-in sensor attached to an OpenPnP Actuator. See some examples [here](https://duckduckgo.com/?q=touch+plate+CNC&t=ffab&iar=images&iax=images&ia=images). 
- Using a passive block to probe part heights using a ContactProbeNozzle. 

_Mark