openpnp/openpnp #99: Visual Navigation Panel

vonnieda, 2015-06-23:
A visual navigation panel which allows the user to easily navigate through the various objects involved in the setup and operation of OpenPnP.

This panel is intended to become the primary UI in the system. It will support machine movement operations and provide a jumping off point for configuration tasks.

vonnieda, 2015-06-23:
Original notes on this feature:

OpenPnP Nav Panel

Development Plan
---
* Be able to easily navigate (via code) to the Board Locations, Placements and Feeders. Later we'll need the same for everything that can be configured (Cameras, Nozzles, Actuators, etc.)
* Think about the current multiple camera views. Is it needed anymore? How will we work on bottom vision setup tasks?
* Need to be able to separate object renderers from the objects. Maybe similar to how we do wizards. Something like getNavRenderer() does tie us to Swing though and we need to think about the future with HTML, JS, SVG, etc. Renderers can get registered in some manner.
* Need to add table extents and soft limits. Maybe they are one and the same. Limits will probably need to be handled at the driver level because of offsets.
* We'll need objects that can be clicked to glow or whatever. More rendering stuff.
* Board rendering is internal to the app - i.e. there is no need for extensions. Feeder rendering will need to be extensible but we should just provide a basic render for a feeder until extensions are in. It can just be a square with a crosshair for the pick location or whatever.
 

Thoughts
---
* Maybe instead of large, complex wizards we instead have property editors that we support default components for which can be overridden. There is probably already a mechanism in Java Beans for this. then we can really focus on a single property at a time:
	* For instance you want to set offsets, that opens a small window that captures from the camera.
* Consider the concept of single task, single focus. This could make the UI touchscreen capable of being run off an embedded PC with a smaller touchscreen.
* Define machine size somewhere.
* Show feeder pick locations, board corners, current nozzle and camera position, etc.
* Layout ideas: Side by side with camera, hidden behind camera (spacebar to swap), small in corner with hover to zoom, etc.
* We should render what we know about each thing. We know the pads that go on a board, so we can render that. If we have a board outline we can render that. We know how wide a tape is, the outline of a part in it, etc. so render all that.
* Make it possible to click objects in the nav panel to go to or edit them. Click a tape to open it's configuration. Click a pad to open a placement. Click a feeder to edit it.
* Essemtech https://www.youtube.com/watch?v=iAxpUgsogsE
* Zoomy stuff would be cool. From overview, click a board to zoom to that board. Click a placement to zoom to it. Click again to show camera to align it. Etc.
* Show a representation of the head, and make it possible to click on camera, nozzle 1, nozzle 2 to select those for movement.
* I like the idea of clicking around in the camera view to jog and then having the extents of the view be 0.01, 0.1, 1.0, 10, 100mm, etc. for moving that distance. It should appear as hover.

vonnieda, 2015-09-27:
Some additional thoughts on this from last night:

* Jogging should all be handled from the camera.
    * Jogging Z could be click and drag up or down.
    * Jogging C is grab any endpoint of crosshair and drag clockwise or counterclockwise. Releasing moves the machine.
* The nav panel should be like Google Maps, showing you a zoomable representation of the machine. As you zoom in, more detail appears, including the camera view as it's size starts to become a valid concern.
    * Example: You start zoomed out and you see 2D representation of the table and wherever the camera is pointing you see a small rectangle of "real" world. As you zoom in on that spot the camera view becomes bigger and overlays a larger part of the panel.
* There should be no reason to jog anything other than the downlooking camera. Everything else can be handled with touchoff and offsets.
    * So, if you are trying to locate a feeder or board, jog the camera to it and then fire a touchoff. The machine moves the nozzle to the camera location, does a touchoff, measures the offsets and then moves the camera back to the location minus the offset.
* Clicking anywhere in the nav panel moves the camera to the clicked position.
    * With this idea there is no need for jog increments at all. If you need to jog far you just flick the scroll wheel to zoom out, then click where you want to go, then zoom in to get accurate.
* It's probably not necessary to have any lists of things. If everything is represented in the nav panel (feeders, pads, placements, boards, etc.) then you just click that thing to work with it. Flyout panels handle the editing.

vonnieda, 2015-09-27:
May have already dismissed this idea, but putting it here for posterity:

Since rendering the nav panel will be fairly complex and we don't want to have to port it all over the place, we could just render it in OpenPnP and stream it to web clients. 

I think this experience would probably suck for twitch things like scrolling, but it might be worth testing and seeing if it works.

PeeJay, 2015-09-27:
Why not just make it a small web app that runs inside a browser window in OpenPnP?

vonnieda, 2015-09-27:
@PeeJay That's a good idea!

cri-s, 2015-09-27:
I think, this is a really bad idea. The performance goes down. Not for the
webserver , but for other reasons.
If it's optional, fine, but if not, VNC or other remote desktop application
probably works better.

vonnieda, 2015-09-29:
Really like some of the color schemes in this screenshot. Need to learn how to add noise to brushes to add a little depth.

[image]

FlyingLotus1983, 2015-09-29:
:+1:  :+1:  two thumbs way up on that.

But what do you mean by adding noise to brushes?

vonnieda, 2015-09-29:
It's a designer trick to add depth to solid colors. If you zoom in on that screenshot you'll see that none of the colors are just solid colors, they all have some amount of semi-transparent noise or a gradient. You'll notice the same thing being used on any website or app that looks nice.

Specific to my comment, everything in the nav panel is drawn with simple 2D primitives. It all looks very flat and full because of the lack of noise, so I wanted to figure out how to add noise to the 2D primitives.

vonnieda, 2016-03-29:
Idea: Whenever the head is idle for more than the camera's settle time, grab an image and tile it into the background. This will produce a visual map of the machine over time.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

This has been a pet issue of mine for years, but no one has ever shown any interest in it, so no reason to keep this open.

closed 2017-04-13 by vonnieda