openpnp/openpnp #106: Gcode Driver With Loadable Profiles

vonnieda, 2015-06-24:
This is an idea that has been percolating for a while. Better write it down. From a recent email on the topic:

A combined driver is on my TODO list, but I had intended to take a slightly different tack. I was thinking that the driver would be generic and just read the proper gcode to use from a profile in the machine.xml. This way people could easily control any gcode based machine. I had intended to make the profiles exportable and importable so that people can just distribute the profile for a given machine or controller.

You can see the start of this in the current version of the OpenPnP reference MarlinDriver (not the FirePick one). Most of the gcode there is loaded from the machine.xml.

There are a couple things that would need to be a bit dynamic - error message formats, response formats, etc. but I think those can be handled pretty simply with a regex.

vonnieda, 2015-06-24:
The first step here should be to go through each of the existing drivers and document the gcode that is currently used. This can be used to build the profiles after the code is finished.

PeeJay, 2015-06-24:
I've made a start on this - see https://github.com/PeeJay/openpnp/blob/GcodeDriver/src/main/java/org/openpnp/machine/reference/driver/GcodeDriver.java

FlyingLotus1983, 2015-06-24:
I've got a gcode abstraction later that j started a while back. I could
share it with you if its helpful?

vonnieda, 2015-06-24:
@PeeJay Looks pretty good so far, although I'm not sure about the driverFirmware idea. I had thought we would just have gcode and regexes in the profile that would allow things like "Say hello to firmware" and "expected response from hello".

PeeJay, 2015-06-25:
Just thinking about this some more - Do we really want each individual user to have their own set of Gcode in the settings file? And regexs? eek! That sounds like a pain to support. Would it be better to have a master file that comes with the openpnp distribution that has one set of settings? I guess long term users won't be looking in machine.xml anyway. Just some thoughts.

PeeJay, 2015-06-25:
@FlyingLotus1983 Sure I'll take all the help I can get.

FlyingLotus1983, 2015-06-25:
@PeeJay sent you the google docs link...

vonnieda, 2015-06-25:
@PeeJay That's what I mean about "loadable profiles". Look at my original comments at the start of the issue. Basically, I see this working like all of the major 3D printing controllers do. You have a file you distribute that contains the profile for a machine. The user loads that file into their machine and done. Doesn't really matter where it's stored in the end, but having it in the machine.xml keeps it tidy and makes it easy to debug if I need to ask someone to send me their config.

The machine.xml is not really intended for users to edit. Over time everything is exposed in the UI. Yes, if they dig deep enough they'd be able to find the gcodes and regexes, but this is how things like Simplify3D and Cura work too. I think it's fine.

PeeJay, 2015-06-25:
@vonnieda Ahh I get it now. So if a profile potentially got corrupted we just tell the user to reload profile X and don't mess with it. I'll keep working on it.

@FlyingLotus1983 Thanks for that. FYI it's quite similar to the list at http://reprap.org/wiki/G-code .

vonnieda, 2015-06-25:
@PeeJay Yep, exactly! The key is that it's a separate file from the machine.xml and that it can be loaded in. That way if you DO need to reload it, or maybe there is a firmware update that requires changes, you don't have to overwrite your entire config containing all your settings, parameters, feeders, etc. You just load in the machine profile file which overwrites the gcode settings.

vonnieda, 2016-01-07:
Also see #121

vonnieda, 2016-04-02:
First version of three has been released. Information at https://github.com/openpnp/openpnp/wiki/GcodeDriver

vonnieda, 2016-05-23:
Also see #273

vonnieda, 2016-06-22:
Release 2 has been completed. More information at https://github.com/openpnp/openpnp/wiki/GcodeDriver#mapping-axes-to-headmountables

mfarahaniasl, 2017-02-02:
HI. How can i find GcodeDriver PCB project or buy one of them?

vonnieda, 2017-02-02:
@mfarahaniasl GcodeDriver is a software driver for connecting OpenPnP with any type of Gcode based controller board. There is no GcodeDriver PCB. I recommend the Smoothieboard for a controller: http://smoothieware.org/getting-smoothieboard

mfarahaniasl, 2017-02-02:
Thank for your replay. There are any openSource Driver board that can work with OpenPnP project too?

vonnieda, 2017-02-02:
@mfarahaniasl Yes, the SmoothieBoard I mentioned is completely open source. Hardware and firmware. You can also use TinyG which is open source.

PeeJay, 2018-06-27:
I'm looking at adding tcp support to the gcode driver, but need a bit of guidance. I'm using @anthonywebb 's tcp driver at https://github.com/anthonywebb/openpnp, which replaces the AbstractSerialPortDriver class with a tcp version. The part I'm unsure of is how to use both classes. 
Normally in C++ I would just instantiate either the serial or tcp class as needed from within the gcodeDriver class, but to do that I would need to make them a full class rather than abstract, and the GcodeDriver would not inherit them any more. Am I going in the right direction here? Java is all a bit new to me.

vonnieda, 2018-06-27:
Yep, my thought on how this would work is that we would have a base
GcodePort class and there would be GcodeSerialPort subclass and
GcodeTcpPort subclass. GcodeDriver would just have an instance of one of
these depending on how it’s configured, or, to make it easier, just have
instances of both and activate the one that’s configured.

vonnieda, 2018-07-21:
For anyone still following along, we now have TCP/IP support thanks to @PeeJay in PR #749. Thanks @PeeJay!

markmaker, 2021-04-25:
I think most of this is covered by Advanced Motion Control and Issues & Solutions smart G-code generation, or made obsolete (it does not make much sense to export/import profiles, if you can generate them automatically).

vonnieda, 2021-04-25:
> I think most of this is covered by Advanced Motion Control and Issues & Solutions smart G-code generation, or made obsolete (it does not make much sense to export/import profiles, if you can generate them automatically).

Agreed - thanks. Closing this.

closed 2021-04-25 by vonnieda