openpnp/openpnp #118: Manual Nozzle Tip Changer

vonnieda, 2015-06-27:
We need to have the ability to signal the operator that a nozzle tip change is needed instead of only supporting automatic changes.

vonnieda, 2016-01-19:
See #183. Adding a NozzleTipChanger interface would be part of that change and would enable this one.

vonnieda, 2016-06-06:
Note that with the recent changes to the JobProcessor it is now possible to work through this manually. We still need to do the work in this ticket to make it a good user experience, but currently if there are no nozzle tips found to handle a part the job will pause and you can make manual adjustments before continuing.

netzmark, 2017-11-13:
Hi Jason,
This is Marek. Here are the changes that I've made with Cri's help to realize this function. It works ok, we use it for everyday. Unfortunately I don't have Auto-Changer station to check it responsible with its too, pls do. However looking on logged codes it seems to be ok.

ReferenceNozzle.java:
from @300    
        // if (changerEnabled) {
            double speed = getHead().getMachine().getSpeed();
            Logger.debug("{}.unloadNozzleTip(): Start", getName());
            ReferenceNozzleTip nt = (ReferenceNozzleTip) nozzleTip;
            Logger.debug("{}.unloadNozzleTip(): moveTo End Location", getName());
            MovableUtils.moveToLocationAtSafeZ(this, nt.getChangerEndLocation(), speed);

//Sending the head to Changer end-location position when manual changing required
	if (changerEnabled) {
        Logger.debug("{}.unloadNozzleTip(): moveTo Mid Location 2", getName());
        moveTo(nt.getChangerMidLocation2(), nt.getChangerMid2ToEndSpeed() * speed);
        Logger.debug("{}.unloadNozzleTip(): moveTo Mid Location", getName());
        moveTo(nt.getChangerMidLocation(), nt.getChangerMidToMid2Speed() * speed);
            Logger.debug("{}.unloadNozzleTip(): moveTo Start Location", getName());
            moveTo(nt.getChangerStartLocation(), nt.getChangerStartToMidSpeed() * speed);
            moveToSafeZ(getHead().getMachine().getSpeed());
            Logger.debug("{}.unloadNozzleTip(): Finished", getName());
        }
        nozzleTip = null;
        currentNozzleTipId = null;
        firePropertyChange("nozzleTip", null, getNozzleTip());
        ((ReferenceMachine) head.getMachine()).fireMachineHeadActivity(head);
// To get pause and exception for manual nozzle change pause
        if (!changerEnabled) {
	throw new Exception("Manual NozzleTip change required!");
	}
    }
    @Override
    public Location getLocation() {
.........

ReferencePnpJobProcessor.java:
From line @499
            NozzleTip nozzleTip = findNozzleTip(nozzle, part);
		//status line text for Manual nozzle changing			
			fireTextStatus("Change NozzleTip on Nozzle %s to %s.", 
				nozzle.getId(), nozzleTip.getName());			
 Logger.debug("Change nozzle tip on {} from {} to {}",
.........

vonnieda, 2017-11-21:
Fixed in #526, thanks @netzmark!

closed 2017-11-21 by vonnieda