openpnp/openpnp #120: Bug: receives error when starting job

la3pna, 2015-06-29:
When starting a job, some times the following errorbox appears as shown in picture file ( and: http://imgur.com/atS9pjZ)

[image]

Packages are allowed for the particular nozzle and "Allow Incompatible packages" is marked. 

machine.xml, job and board, and packages and parts are sendt Jason on e-mail, since github don't like that kind of files uploaded to issues?

vonnieda, 2015-06-30:
It looks like this issue is to due to a tray feeder being out of parts (feed count > tray count), but the error message is very deceptive.

vonnieda, 2015-06-30:
Okay, the core issue here is that the planner does not do a good job of indicating what is wrong with a particular solution. If there is a viable nozzle but no viable feeder, we end up returning the original, unplanned solution which has nozzle and feeder set to null. In the processor code, we just happen to check for null nozzle before feeder, so even though this is a feeder issue the error we show is that the nozzle is null.

There is a TODO in the code regarding this, at https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/planner/SimpleJobPlanner.java#L81

As mentioned in the TODO, we either need to go ahead and plan a partial solution so the processor can get a reasonable idea of the problem, or just throw an error during planning. Not sure what the best option is just yet, so putting there here for rumination.

vonnieda, 2015-06-30:
To reproduce the issue, create a tray feeder and set it's feed count > tray count and attempt to start a job that references it.

commit f2e3aa40 (2016-02-15): SimpleJobPlanner will now return partial solutions when no viable feeder or nozzle tip is found. This allows the processor to show the correct error message in these conditions. This is the first step in fixing several issues. This fixes #120, specifically with more to come.
closed 2016-02-15 by vonnieda