openpnp/openpnp #124: opencv, videoinput lib missing

cri-s, 2015-07-03:
Hi, the videoinput lib is missing, same as ffmpeg .
configure WITH_VIDEOINPUT
Having videoinput and relative java wrapper allow on windows plattform getting the list of cameras
and names using this pice of code:
public static void main( String args[] )
  {  
    int numDevs = videoInput.listDevices();
    System.out.println("No of video input devices: " + numDevs);
    for (int i = 0; i < numDevs; i++)
      System.out.println(" " + i + ": " + videoInput.getDeviceName(i));
  }  // end of main()

For mac and linux it is possible to get the list too without problems, but if windows is not covered,
then adding code to output list of cameras don't make sense.

ffmeg is required by some cameras to decode the encoded mpeg stream, that require less bandwith than yuv getting more fps and allowing more  cameras on the same usb root controller or hub.

vonnieda, 2015-07-04:
@cri-s can you expand on "For mac and linux it is possible to get the list too without problems"? Is this something you are doing in OpenCV, or externally?

Are you suggesting that we use the videoinput library to get the camera list on windows and use other methods on Mac and Linux? I think this would be a good idea, if we can make it work, since it doesn't seem like OpenCV is going to add camera enumeration anytime soon.

Related to #73

cri-s, 2015-07-04:
>
> @cri-s can you expand on "For mac and linux it is possible to get the
list too without problems"? Is this something you are doing in OpenCV, or
externally?

Externally.
>
> Are you suggesting that we use the videoinput library to get the camera
list on windows and use other methods on Mac and Linux?
Yes.

I think this would be a good idea, if we can make it work, since it doesn't
seem like OpenCV is going to add camera enumeration anytime soon.

FlyingLotus1983, 2015-07-04:
Assuming linux and mac have to be handled externally to enumerate cameras, then Libwebcam would be one option, for linux and possibly mac:

http://sourceforge.net/projects/libwebcam/

It allows enumeration of cameras, and allows dynamic control of brightness, exposure, and other settings via uvcdynctrl.

I'm not aware of a Java binding for it, but it's not impossible to come up with one.  This has actually been on my list for some time to investigate.  I know some of our competitors are using it with great results so far, on Linux.

vonnieda, 2015-07-04:
As a note to myself (or others working on this), we can either try to shoehorn this kind of functionality into the OpenCvCamera or add a new camera type that handles it. It's also worth considering just using the capture functionality of another library since OpenCV has proven not to be very good for that.

cri-s, 2015-07-04:
On linux, you just need to read the dir /dev/v4l/by-id/
Further it is possibile using udev rules to always assign the same id
indipendently of where
the usb cam are plugged in, if required.

cri-s, 2015-07-18:
Win and Linux works, for Mac, proof of concept was done and tested, but further test is needed and 
this is actually missing, so actually i close this issue because it's seems there is not forthcoming on Mac
front.

closed 2015-07-18 by cri-s