openpnp/openpnp #125: segfault on linux , cv::initInterTab2D bug

cri-s, 2015-07-03:
actually there is a segfault on linux,  cv::initInterTab2D, seems a gcc bug.
The cpu have the sse2 flag.

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x9ed585cc, pid=2834, tid=2685029184
#
# JRE version: Java(TM) SE Runtime Environment (7.0_80-b15) (build 1.7.0_80-b15)
# Java VM: Java HotSpot(TM) Client VM (24.80-b11 mixed mode linux-x86 )
# Problematic frame:
# C  [libopencv_java249.so+0x50a5cc]  cv::initInterTab2D(int, bool)+0x2fc
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  T H R E A D  ---------------

Current thread (0xa01a5c00):  JavaThread "AWT-EventQueue-0" [_thread_in_native, id=2847, stack(0xa0054000,0xa00a5000)]

siginfo:si_signo=SIGILL: si_errno=0, si_code=2 (ILL_ILLOPN), si_addr=0x9ed585cc

Registers:
EAX=0x00000000, EBX=0x9f74dff4, ECX=0xb6b61000, EDX=0x9f757b50
ESP=0xa009fdd0, EBP=0x00000001, ESI=0xa009feec, EDI=0x9f76e768
EIP=0x9ed585cc, EFLAGS=0x00010286, CR2=0x00000000

Top of Stack: (sp=0xa009fdd0)
0xa009fdd0:   75dbaa2c c0030bc0 a009feb8 a009feb0
0xa009fde0:   d4a23adb 00000000 1653223e 00000000
0xa009fdf0:   00000000 00000000 00000000 00000003
0xa009fe00:   40420000 a88ec105 00000000 00000000
0xa009fe10:   80000000 00000000 00000000 00000000
0xa009fe20:   115bc42a 00000000 1835084e a009fecc
0xa009fe30:   00020774 402fb4e2 3d000000 a00a02cc
0xa009fe40:   00000002 00000008 a009fecc 00000001 

Instructions: (pc=0x9ed585cc)
0x9ed585ac:   c2 10 0f 14 c0 0f 5a c0 f2 0f 59 83 ac 42 cf ff
0x9ed585bc:   f2 0f 59 83 24 2c cf ff f2 0f 59 c0 f2 0f 5e c8
0x9ed585cc:   f2 0f 12 c9 66 0f 5a c9 f3 0f 11 0c 86 83 c0 01
0x9ed585dc:   f3 0f 58 d9 83 f8 08 75 a0 f3 0f 10 83 ac d0 ce 

Register to memory mapping:

EAX=0x00000000 is an unknown value
EBX=0x9f74dff4: <offset 0xeffff4> in /tmp/3626301020705460178/nu/pattern/opencv/linux/x86_32/libopencv_java249.so at 0x9e84e000
ECX=0xb6b61000: <offset 0x45000> in /lib/i386-linux-gnu/libm.so.6 at 0xb6b1c000
EDX=0x9f757b50: <offset 0xf09b50> in /tmp/3626301020705460178/nu/pattern/opencv/linux/x86_32/libopencv_java249.so at 0x9e84e000
ESP=0xa009fdd0 is pointing into the stack for thread: 0xa01a5c00
EBP=0x00000001 is an unknown value
ESI=0xa009feec is pointing into the stack for thread: 0xa01a5c00
EDI=0x9f76e768: <offset 0xf20768> in /tmp/3626301020705460178/nu/pattern/opencv/linux/x86_32/libopencv_java249.so at 0x9e84e000

Stack: [0xa0054000,0xa00a5000],  sp=0xa009fdd0,  free space=303k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libopencv_java249.so+0x50a5cc]  cv::initInterTab2D(int, bool)+0x2fc

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  java.lang.ClassLoader$NativeLibrary.load(Ljava/lang/String;)V+0
j  java.lang.ClassLoader.loadLibrary1(Ljava/lang/Class;Ljava/io/File;)Z+302
j  java.lang.ClassLoader.loadLibrary0(Ljava/lang/Class;Ljava/io/File;)Z+2
j  java.lang.ClassLoader.loadLibrary(Ljava/lang/Class;Ljava/lang/String;Z)V+268
j  java.lang.Runtime.loadLibrary0(Ljava/lang/Class;Ljava/lang/String;)V+54
j  java.lang.System.loadLibrary(Ljava/lang/String;)V+7
[... 629 lines cut ...]
siblings	: 1
core id		: 0
cpu cores	: 1
apicid		: 0
initial apicid	: 0
fdiv_bug	: no
f00f_bug	: no
coma_bug	: no
fpu		: yes
fpu_exception	: yes
cpuid level	: 2
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov clflush dts acpi mmx fxsr sse sse2 ss tm pbe nx bts est tm2
bogomips	: 3399.92
clflush size	: 64
cache_alignment	: 64
address sizes	: 32 bits physical, 32 bits virtual
power management:

Memory: 4k page, physical 1009596k(112936k free), swap 2370556k(2352924k free)

vm_info: Java HotSpot(TM) Client VM (24.80-b11) for linux-x86 JRE (1.7.0_80-b15), built on Apr 10 2015 12:41:05 by "java_re" with gcc 4.3.0 20080428 (Red Hat 4.3.0-8)

time: Fri Jul  3 18:02:38 2015
elapsed time: 4 seconds

vonnieda, 2015-07-04:
http://code.opencv.org/issues/3163 could be related.

vonnieda, 2015-07-04:
@cri-s what function is calling this to trigger the segfault? Is this something already in the OpenPnP codebase or something you've added? Can you provide a test case or a description of how to reproduce?

cri-s, 2015-07-04:
I have used the opencv libs from latest devel or master codebase on 32bit
OS that get downloaded automatically on mvn package. It seems this is GCC
bug and
If using newer GCC it should gone.
Using older opencv libs from previous openpnp release works without
problems.
Actually at least 32bit linux is actually broken.
Further it need CPU with sse2 extension.
I have it, but its's not granted that this is given, and such chance of
requirement should be documented.

Its the plain opencv and system is up to date with updates. After starting,
it crashes immediatly.

cri-s, 2015-07-08:
removed 2.4.9-7 and used 2.4.8, removed nu.pattern call and that works for me.
Require opencv is installed on system.

closed 2015-07-08 by cri-s