openpnp/openpnp #126: enhancement TubeFeeder

la3pna, 2015-07-04:
* TubeFeeder should allow actuator choice. Preferably with a way to set G-code for this actuator.

* Actuator should be set on for a given time. This could be before or after the part is picked up. perhaps this can be set by having the time positive for before the part is picked, and negative for after, or a checkbox. 

*  Vision could be used to check if component are available or run the feeder another duration.

*  Can vision be used to determine if the component are at 0 degrees or 180 degrees? some manufacturers seems to not care what direction the parts are in the tube.

cri-s, 2015-07-04:
>
> TubeFeeder should allow actuator choice. Preferably with a way to set
G-code for this actuator.

Why, just curios about the usage case.

>
> Actuator should be set on for a given time. This could be before or after
the part is picked up. perhaps this can be set by having the time positive
for before the part is picked, and negative for after, or a checkbox.

Having actor after picking can gives some problems on parts.

> Vision could be used to check if component are available or run the
feeder another duration.
If vision is enabled yes.

>
> Can vision be used to determine if the component are at 0 degrees or 180
degrees? some manufacturers seems to not care what direction the parts are
in the tube.
Yes, if making feeder group and two differenti templare, and same feeder
pos. I can give the code.

FlyingLotus1983, 2015-07-04:
This is a moot point, until someone had a working vibe tube feeder to
integrate into the machine.

And furthermore, many of the mid-grade and high end machines I've used just
leave the vibe on all the time, it doesn't need to be turned off. and in
fact, many of them are ganged to several parts (4 or 5 tubes get vibrated
by a single motor), which is complicated and trick, albeit not impossible
to write a driver for.

Priority wise, (not speaking for anyone here, just my own opinion), this is
at the bottom of the list, until someone has at least made plans or
commercially offering some sort of tube feeder in which the driver needs to
be written.

-- 
Neil Jansen, Co-Founder
Tin Whiskers Technology, LLC
http://tinwhiskers.io

FlyingLotus1983, 2015-07-04:
On Saturday, July 4, 2015, Thomas S. Knutsen <notifications@github.com>
wrote:

>
>    -
>
>    Can vision be used to determine if the component are at 0 degrees or
>    180 degrees? some manufacturers seems to not care what direction the parts
>    are in the tube.
>
> This is just not true in my experience. Can you provide any concrete
evidence of a manufacturer documenting that they sell parts randomly placed
into tubes?  I used to run an SMT line, for years, and every tube that I
have ever loaded has had every single part come out with pin one in the
same direction.

-- 
Neil Jansen, Co-Founder
Tin Whiskers Technology, LLC
http://tinwhiskers.io

vonnieda, 2015-07-04:
This all sounds fine to me, except one thing. The actuator gcode would be handled at the driver level, not in the actuator or feeder setup. The actuator system is already independent of the rest of the system and will remain that way. It's easy, though. Just have an index on the actuator, or a name, or anything else and the driver can use that to determine which gcode it will send.

la3pna, 2015-07-04:
I have made the parts for a simple tubefeeder that holds 5 tubes, and
vibrates them. The feeder is made for PLCC44 size with adapters for the
smaller sizes. It holds 5 tubes and vibration is from 2 cellphone
vibrators.
The parts I have made are milled in acrylic, but there is no reason that 3d
printing should be a problem, so it should be easy to reproduce. Plan to
publish the drawings over the next couple of weeks, just need to tidy them
up a bit.

As for priority, I agree that this goes down on the list, I just wanted to
add it to reference some discussion we have had about it in the IRC channel.

la3pna, 2015-07-04:
> evidence of a manufacturer documenting that they sell parts randomly placed
> into tubes? I used to run an SMT line, for years, and every tube that I
> have ever loaded has had every single part come [...]

Nope, just had a look over the parts I have here. Seems that several of my
tubes of National LMX series PLL's have their parts randomly placed, but
this can be that the store I used have repackaged or dropped them.
Having vision check for the orientation should also fix the good old "I
inserted the tube the wrong way again" problem...

FlyingLotus1983, 2015-07-04:
Do you have a datasheet for that specific part?  More than likely the part orientation will be clearly listed in the datasheet.  If you buy these parts from a reputable supplier (Digi-key, Mouser, Arrow, etc), they would come with pin 1 in the same direction.  If you buy from someone else, it's possible that they get re-packaged and moved around.

Doing vision for this sort of stuff is not trivial (recognizing random parts that the machine has never seen before, recognizing their orientation with all sorts of variables like light level, reflection, camera differences, etc).  I think we wall want it in the end, but it's actually an incredibly hard problem to solve.  It's a problem I plan on personally working on this year, but I don' t think it will be ready for quite some time.

cri-s, 2015-07-05:
I have vibrating feeder too.
The higher end one. Vibrating 220V with
Dimmer and motor with speed regulation
That let oscillate the carrige 8cm.

The one at the fair have used modified air pump for vibrating, i have used
electric hair cutter.

In the remote case i want pause it when not needed i would add IR reflex
sensor or capacitive/inductive self oscillating sensors and override switch
in order to pause instead of timer.
Yes i can do the vision stuff but i have the limitation that feeder id can
only be numeric.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

closed 2017-04-13 by vonnieda