openpnp/openpnp #127: One camera active at a time support

vonnieda, 2015-07-05:
Due to hardware limitations it is sometimes not possible to have two USB cameras active at once. There are several threads on the OpenPnP and Firepick lists about this. The end result is that we need a way to only activate a camera when it is needed. and to turn off the others before we use it. This can probably be done with a camera wrapper class that all of the cameras in a machine use that talk to each other to coordinate.

cri-s, 2015-07-05:
Im actually doing that.

FlyingLotus1983, 2015-07-05:
:+1:  this seems like a great approach to the problem.  I do think it might be wise to get some metrics of how many milliseconds each camera needs to start and stop, because if we needed to do this constantly, it might delay the job if it took too long.  I'd say anything under ~250 ms to transition should be perfectly fine.

Jason, I think the machine processor might be a good place to call the enabling and disabling of the cameras.  That way different machine processors could control how the cameras interacted (rather than just mandating it happen one way or another).  If you can imagine a few scenarios of certain machines wanting to use cameras simultaneously, and others only needing them in a serial process, the processor seems like the best place to allow this behavior to be implemented (e.g. call the camera enabling / disabling).

vonnieda, 2015-07-05:
Hey @FlyingLotus1983, I was actually thinking of making this a bit simpler than that. My thought was to have something like a SingletonCameraWrapper. It would implement Camera and wrap an existing Camera instance and then it would manage synchronizing the Camera access. This way the feature is entirely contained by the Cameras and doesn't bleed into the rest of the system. People who need it could add it and others just don't.

machine.xml would look something like:

```
<cameras>
    <camera class="org.openpnp.reference.camera.SingletonCameraWrapper" group="0">
        <camera class="org.openpnp.reference.camera.OpenCvCamera" device-id="0">
        </camera>
    </camera>
    <camera class="org.openpnp.reference.camera.SingletonCameraWrapper" group="0">
        <camera class="org.openpnp.reference.camera.OpenCvCamera" device-id="1">
        </camera>
    </camera>
</cameras>
```

The SingletonCameraWrapper would have a singleton internal class responsible for managing which camera was on at any time for a given group.

I agree with collecting some metrics on this. I know that Camera startup is one of the slowest parts of startup in OpenPnP currently.

FlyingLotus1983, 2015-07-05:
@vonnieda Re: metrics, easiest way might be to just add some debug logging into the main/latest 'dev' branch, that reports how long it took to initialize and close the camera drivers... Let a few people run this and report back with how long various cameras took.  I have probably 5-8 cameras here I could try, and can try on 3 OS's.  As long as it gets reported in log4j it should be easy to try a few, before any real work is started.

vonnieda, 2015-07-05:
Note from Firepick mailing list, which may come in helpful when diagnosing multiple camera support issues:

Can't speak for Windows, but I was unable to use a hub with both cameras to work on a Mac or on Linux.
The cameras much each be connected to a different USB bus. Under Linux, you can query the USB buses and get their current bandwidth utility.
On my PC:
root@Den:/home/shmuelzon# cat /sys/kernel/debug/usb/devices | grep "B:"
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
B:  Alloc= 30/900 us ( 3%), #Int=  1, #Iso=  0
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
B:  Alloc=  0/900 us ( 0%), #Int=  0, #Iso=  0
B:  Alloc= 25/900 us ( 3%), #Int=  3, #Iso=  0
B:  Alloc=488/800 us (61%), #Int=  2, #Iso=  5
B:  Alloc=488/800 us (61%), #Int=  7, #Iso=  5

The last two are the busses I have the camera's connected to and are currently viewed via VLC. Since the both take 61% of the USB2 bus they won't work simultaneously on the same bus.
I'll also note that I tried having a single camera connected to a cheap-o USB2 hub along with the Arduino and with that, the camera didn't work either. I then connected it to a branded USB3 hub (even though the PC port is USB2) and that did work.

cri-s, 2015-07-05:
i have measured my cameras with wireshark.
The downlooking 640x480 camera 25fps [mjpeg] have bandwidth of 68 Mbit/sec.
The uplooking 1280 x 720 camera 30fps [mjpeg] have bandwidth of 20
Mbit/sec , worst case 120 Mbit/sec. Compression is extremly good on
this cam.
On all audio was not activate, otherwise the audio increase the
bandwidth 100 Mbit/sec/cam
The second uplooking cam is the same as downlooking one. All three
operate at the same
time being on the same hub and single usb 2.0 port.

zrecommerce, 2015-07-06:
I would *strongly* urge developers to consider keeping your camera routines
asynchronous to the rest of your application - and suggest cameras be
hooked up to externally powered USB ports on machines that may not have
enough juice on their older USB ports.

I've learned USB cameras will eventually lose USB power due to Windows 7
"Power Saving" profiles. They may not power up quickly. Even when disabling
the power saving feature on the USB lines, USB cameras may still power off
after a while (not sure why). Switching between cameras may become an issue
because of slow or non-power up.

My 2 cents. Hope it helps.

Alex Albino
Sr. Fullstack Developer
Zend Certified Engineer
https://github.com/zrecommerce
https://github.com/femtoduino
https://github.com/femtoio

(707) 246-6864

cri-s, 2015-07-07:
Here test case.
If running without arguments, it does front face detection.
If running with argument, just the argument count is checked, it
display the webcam and
run faster. TSC is the cpu performance timer counter. On certain
systems is completly unreliable, so both TSC and traditional timing is
used. It is possible to restart the same camera and time that. There
is no check how many cameras are present.

After 20 frames, fps is calculated.

vonnieda, 2015-07-07:
@cri-s Did you mean to attach something?

cri-s, 2015-07-07:
If pnp doing downlooking and have place for placing temporary some chips and job processor sort
for camera usage, process all placement without camera and with downlooking camera, then move
the chips that use the downlooking and uplooking camera to temporary tray feeder, switch camera and 
then doing uplooking camera and pick the components from temporary tray feeder and perform 
uplooking processing, why not. On my system, timing is from 1.5 to 3 seconds doing full framerate and from 2.5 to 16 seconds with 2 fps, depending on camera, one does from 2.5 to 10.5 seconds,
the other need 3.3-16 seconds and it's really wired how often it is fast and the really slow. 
I suspect the cameras that have this problem as it don't have compression are faster.

cri-s, 2015-07-07:
yes, i have respond using email and attached something. I have send it separatly after on mail to
@FlyingLotus1983 and @vonnieda

cri-s, 2015-07-10:
Do you have any result to share, i'm curious about real switching time on cameras without compression, on both, low and higher fps including the fps readout info.
If you have problems with java classpath, tell it me and i pack it to jar.

vonnieda, 2016-01-13:
Closing this as "wont fix". The libuvc camera will be released soon and is a better solution to this problem. In the meantime users can work around by using multiple USB host controller ports.

closed 2016-01-13 by vonnieda