openpnp/openpnp #128: Step and Repeat for PCB Panels

vonnieda, 2015-07-08:
A common pick and place use case is placing a panel of one or more of the same board. Currently to set this up you'd need to create a job with multiple copies of the board and set the position of each. Since the positions of boards in a panel are well known it should be possible to create the job simply by specifying an X and Y count and offset and OpenPnP would duplicate the board that many times and set the position based on the offsets.

A caveat is that if the panel is moved on the table the positions of each board will have to be reset. It should be possible to just set the position of each as above, even after initial job creation.

Another option would be to add specific handling for panels, where the job would contain data about the panel instead of just duplicating the boards. This would be better in the long run but could be done as a phase 2 of this feature. Implementing it as above would be very simple while adding panel data would be more complex.

zrecommerce, 2015-07-08:
Can the panelized job use fiducials? Perhaps fiducials should be strongly
recommended.

I know that with other PnP solutions (such as LitePlacer), the component
list must contains three (3) or more fiducials. In combination with setting
a PCB "Zero" location, the top camera is used to calculate the exact
component locations (since there may be slight unsquareness with the frame)

Alex Albino
Sr. Fullstack Developer
Zend Certified Engineer
https://github.com/zrecommerce
https://github.com/femtoduino
https://github.com/femtoio

(707) 246-6864

vonnieda, 2015-07-08:
@zrecommerce Yep, fiducials are supported in jobs currently, as is checking and updating positions based on them during job start. You just check the "Check Fids?" checkbox for each board.

mikeanton, 2015-09-08:
How about adding a base offset that is applied to everything?  On one machine I used, this was called the bias point, and it also offset the locations of the individual board in a panel.

cri-s, 2015-09-08:
I use panel fiducials and offsets.
The board rotation and offsets have a queue instead of variable that is
used.

cri-s, 2015-09-08:
I have checked just the official code (develop) if i could submit the
patch to add queue instead
of variable, but it's not possible. The fiducial code seems to ignore
completely rotation and
further it updates board position, thing that don't should be done. I
could be wrong with
such code review.

2015-09-08 12:29 GMT+02:00, Cri S <phone.cri@gmail.com>:
> I use panel fiducials and offsets.
> The board rotation and offsets have a queue instead of variable that is
> used.
>

vonnieda, 2015-09-08:
@cri-s can you explain what you mean by "The fiducial code seems to ignore completely rotation"? Do you mean board rotation, fiducial rotation or other? The fiducial code is designed to find and correct X/Y offset and rotation and works well. If you are seeing otherwise, please let me know how to reproduce.

cri-s, 2015-09-08:
@jason, i tray to explain shortly what i have seen, but take it as
"possible wrong" as
review was made too hurry.

The code seems trick and need further investigation, hovever:
Code from two board fiducial code:

// this returns offsets and Angle
 Location boardLocation = Utils2D.calculateAngleAndOffset(
                        placementALocation,
                        placementBLocation,
                        visionA,
                        visionB);

// adjust units, if needed.
                Location oldBoardLocation =
jobPanel.getSelectedBoardLocation().getLocation();
                oldBoardLocation =
oldBoardLocation.convertToUnits(boardLocation.getUnits());

// update Z position to
                boardLocation = boardLocation.derive(null, null,
oldBoardLocation.getZ(), null);

// update board selection
                jobPanel.getSelectedBoardLocation().setLocation(boardLocation);
                jobPanel.refreshSelectedBoardRow();

It should be handled different and not update board location from my
point of view.
Now code for calculateAngleAndOffset is follow:

    public static Location calculateAngleAndOffset(Location idealA,
Location idealB, Location actualA, Location actualB) {
        idealB = idealB.convertToUnits(idealA.getUnits());
        actualA = actualA.convertToUnits(idealA.getUnits());
        actualB = actualB.convertToUnits(idealA.getUnits());

        double angle = Math.toDegrees(Math.atan2(actualA.getY() -
actualB.getY(), actualA.getX() - actualB.getX())
                - Math.atan2(idealA.getY() - idealB.getY(),
idealA.getX() - idealB.getX()));

        Location idealARotated = idealA.rotateXy(angle);

        Location offset = actualA.subtract(idealARotated);

        return new Location(idealA.getUnits(), offset.getX(),
offset.getY(), 0, angle);
    }
}

This function is a bit strange as it don't return Offset and Angle
difference, but in reality
it returned updated X/Y position and angle difference between measured
and given coordinates.
So reviewing the above code, it seems that the resulting Angle from
board location is wrong, but it don't matter if it was Zero, even
possibly the direction of rotation could be wrong, i don't have
investigated that.

Checking not four-points code, but i don't have checked it in depht as i should:
   Location boardLocation = new Location(Configuration.get()
                                .getSystemUnits(), dx, dy, 0, angle );

So dx and dy should have board X and Y coordinate.
both are defined as:

 double dx = dxMP;
 double dy = dyMP;

and dxMP is defined as:

                     dA = (boardLocationA.getY() - A.getY());
                     dB = (boardLocationB.getY() - B.getY());
                     dC = (boardLocationC.getY() - C.getY());
                     dD = (boardLocationD.getY() - D.getY());

                    // Take the average of the four
                    dyMP = (dA + dB + dC + dD)/4;

Theoreof then dy and dx is the offset and not the position, but i'm not sure.
If this is so, setting the board location as offset is even more wrong
for X and Y.

btw Jason, i have big trouble on updating code, maybe you could help
because for you is maybe simple.
On PlacementPartTable, during placement, the Status and Progress
(you'r Type field) don't get updated at all. It get updated, but the
graphical table representation is not updated.
Do you can give indication for that ?

vonnieda, 2015-09-09:
Moving the fiducial discussion and table question to the mailing list as it doesn't pertain to this issue.

issus, 2016-08-06:
Maybe something like Altium's Embedded board array feature:
[image]

I'm not sure how this would handle panels with mixed content (multiple boards in one panel) if you only have centres for components on each board individually, rather than the full panel export - not an issue with Altium but not sure how other CAD packages would export pick and place data.

cri-s, 2016-08-07:
For a request, i have written a script for that.
It should work even without panel fiducial code.

cri-s, 2016-08-07:
because attachement dropped, http://pastebin.com/GAnUEdWw

2016-08-07 18:42 GMT, Cri S <phone.cri@gmail.com>:

vonnieda, 2017-03-06:
PR for this feature is under review: https://github.com/openpnp/openpnp/pull/456

vonnieda, 2017-04-02:
Done in #456, with thanks to @BendRocks!

See https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2017-04-01

closed 2017-04-02 by vonnieda