openpnp/openpnp #132: TinygDriver no longer works with current TinyG code

vonnieda, 2015-07-13:
From the mailing list:

Hi,

So far I've been unable to successfully connect OpenPnP to my TinyG. I've traced through the code a bit to understand the connection sequence but I'm not a Java developer so I'd appreciate any assistance.

From what I can see, the driver starts by requesting the firmware version from the TinyG and checking both the response code for the JSON request and the value of the firmware version itself. Looking in the driver file (src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java) around line 88 is:

JsonObject response = sendCommand("{\"fv\":\"\"}", 500);

That's the "firmware version" command in JSON format, and if I connect to TinyG using a serial terminal and manually type in that command, this is what I see:

{"fv":""}
{"r":{"fv":0.970},"f":[1,0,10,463]}
{"sr":{"posx":0.000,"posy":0.000,"posz":0.000,"posa":0.000,"feed":0.00,"vel":0.00,"unit":1,"coor":1,"dist":0,"frmo":1,"stat":1}}

The "r" response object gives an "fv" value of 0.970, which is correct. Then there is the "f" response value, and an "sr" status report comes along for the ride but can be ignored.

So that's all good: it looks to me like the TinyG responds correctly to the inquiry when I send it manually.

However, I don't think OpenPnP is seeing the same response when it sends that same inquiry.

Jumping down to about line 271, the driver sends the command and looks for both the response value and a response code. Line 287 is:

response = lastResponse;

Immediately after that line I've added the following line, so I can see what it sees:

logger.debug("Last response:" + response);

The output I see in the debug log is:

TinygDriver Last response:{"fv":0.970}

So it looks like the driver is only seeing the fv value in the response, not the entire response. Then a couple of lines later is this:

int responseStatusCode = getResponseStatusCode(response);

We just saw the value of "response" that is being passed to this function call, and I don't think it's correct. I think this function call is expecting the entire response object, including the response code, because this is what the function looks like:

private int getResponseStatusCode(JsonObject o) {
    return o.get("f").getAsJsonArray().get(1).getAsInt();
}

However, the "f" parameter (which you can see in the full response to my manual command further back) isn't included in what has been passed to the function.

Finally, below is the debug log output for one connection attempt. The null pointer exception seems to be caused by the missing "f" value referenced in getResponseStatusCode, as shown above.

I'd appreciate any help with this!

Cheers

Jonathan Oxer

17:21:54,933 DEBUG          TinygDriver sendCommand({"fv":""}, 500)
17:21:54,949 DEBUG          TinygDriver Firmware version check failed
java.lang.NullPointerException
	at org.openpnp.machine.reference.driver.TinygDriver.getResponseStatusCode(TinygDriver.java:242)
	at org.openpnp.machine.reference.driver.TinygDriver.sendCommand(TinygDriver.java:292)
	at org.openpnp.machine.reference.driver.TinygDriver.connect(TinygDriver.java:88)
	at org.openpnp.machine.reference.driver.TinygDriver.setEnabled(TinygDriver.java:136)
	at org.openpnp.machine.reference.ReferenceMachine.setEnabled(ReferenceMachine.java:82)
	at org.openpnp.gui.MachineControlsPanel$8.actionPerformed(MachineControlsPanel.java:466)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2018)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2341)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6516)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3312)
	at java.awt.Component.processEvent(Component.java:6281)
	at java.awt.Container.processEvent(Container.java:2229)
	at java.awt.Component.dispatchEventImpl(Component.java:4872)
	at java.awt.Container.dispatchEventImpl(Container.java:2287)
	at java.awt.Component.dispatchEvent(Component.java:4698)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4832)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4492)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4422)
	at java.awt.Container.dispatchEventImpl(Container.java:2273)
	at java.awt.Window.dispatchEventImpl(Window.java:2719)
	at java.awt.Component.dispatchEvent(Component.java:4698)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:747)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.awt.EventQueue$4.run(EventQueue.java:718)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:717)
	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:349)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:91)
Exception in thread "AWT-EventQueue-0" java.lang.Error: Unable to receive connection response from TinyG. Check your port and baud rate, and that you are running at least version 0.950000 of TinyG
	at org.openpnp.machine.reference.driver.TinygDriver.connect(TinygDriver.java:100)
	at org.openpnp.machine.reference.driver.TinygDriver.setEnabled(TinygDriver.java:136)
	at org.openpnp.machine.reference.ReferenceMachine.setEnabled(ReferenceMachine.java:82)
	at org.openpnp.gui.MachineControlsPanel$8.actionPerformed(MachineControlsPanel.java:466)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2018)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2341)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:252)
	at java.awt.Component.processMouseEvent(Component.java:6516)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3312)
	at java.awt.Component.processEvent(Component.java:6281)
	at java.awt.Container.processEvent(Container.java:2229)
	at java.awt.Component.dispatchEventImpl(Component.java:4872)
	at java.awt.Container.dispatchEventImpl(Container.java:2287)
	at java.awt.Component.dispatchEvent(Component.java:4698)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4832)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4492)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4422)
	at java.awt.Container.dispatchEventImpl(Container.java:2273)
	at java.awt.Window.dispatchEventImpl(Window.java:2719)
	at java.awt.Component.dispatchEvent(Component.java:4698)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:747)
	at java.awt.EventQueue.access$300(EventQueue.java:103)
	at java.awt.EventQueue$3.run(EventQueue.java:706)
	at java.awt.EventQueue$3.run(EventQueue.java:704)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:87)
	at java.awt.EventQueue$4.run(EventQueue.java:720)
	at java.awt.EventQueue$4.run(EventQueue.java:718)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:76)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:717)
	at org.openpnp.gui.MainFrame$2.dispatchEvent(MainFrame.java:349)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:242)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:161)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:150)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:146)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:138)

jonoxer, 2015-07-14:
I can work around the problem by hard-coding the return from getResponseStatusCode() as follows:

jon@jlab:~/src/openpnp#develop$ git diff src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java
diff --git a/src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java b/src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java
index 441e01a..5db9b31 100644
--- a/src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java
+++ b/src/main/java/org/openpnp/machine/reference/driver/TinygDriver.java
@@ -239,7 +239,8 @@ public class TinygDriver extends AbstractSerialPortDriver implements Runnable {
     }

     private int getResponseStatusCode(JsonObject o) {
-        return o.get("f").getAsJsonArray().get(1).getAsInt();
+        //return o.get("f").getAsJsonArray().get(1).getAsInt();
+        return(60);
     }

Obviously this is not a real solution because it fakes the response code from the TinyG, but it localises the problem to that particular line.

cri-s, 2015-08-10:
The problem is as presumed inside run() function, but i have wanted getting the response from log files in order to be sure to have real response line codes.

-                 connectedVersion = response.get("fv").getAsDouble();
+                connectedVersion = response.get("r").getAsJsonObject().get("fv").getAsDouble();

-                     lastResponse = o.get("r").getAsJsonObject();
+                    lastResponse = o;

vonnieda, 2015-08-10:
Can someone test this fix and let me know if it's okay to merge?

jonoxer, 2015-08-10:
YES!

Thanks Cri-s and Jason, I've just compiled and tested this with TinyG 0.97,
and it works perfectly.

Now I can do proper testing without single-stepping through the job and
waiting for each move to complete. Sweet!

Jon

vonnieda, 2015-08-10:
Pull request merged. Thank you Cri S and Jon for sorting this out!

closed 2015-08-10 by vonnieda