openpnp/openpnp #140: Fiducial check should do a sanity check before completing.

vonnieda, 2015-08-17:
Right now the fiducial check can get confused by nearby fiducials or things that look like fiducials. Since it does the two point check independently of each other it's possible to end up with a resulting board orientation that is incorrect.

Suggest:
After fiducial check is complete, check the 2D distance between the found fiducials and the placements and if it's not within a couple percent of each other abort the check or try again.

vonnieda, 2015-08-27:
Checked in a quick mockup of this idea. Checks that the fiducials are within 1% distant of where they should be. Working well so far but won't close this until tested further.

cri-s, 2015-08-27:
Usually error is 0.75% specified by PCB producer. IPC 2221 should be
checked and bow and twist added.
Without checking, I would allow for 1.3%  and if assembling flex PCB or
laser transfered boards for prototype this value need to be increased.

vonnieda, 2015-09-13:
Based on @cri-s comments, we should probably make this a per board setting that the user can configure.

markmaker, 2021-04-25:
This is covered by the #1179 Fiducial locator **Max. Distance** field. See "Fiducial Locator" section.

closed 2021-04-25 by vonnieda