openpnp/openpnp #141: FiducialLocator#createTemplate should not upsize the image.

vonnieda, 2015-08-27:
The current FiducialLocator#createTemplate code creates an image with a 50% border because this produced better matches. The reason this worked is that it created (by accidental) the keepout area that is on a ficudial. While this happens to work for round fids of a certain size it will not work in many cases. It should be changed to generate a template that is specific to the fiducial, either by modifying the footprint to allow for keepout or just specifying a type and size of fiducial.

vonnieda, 2016-01-19:
Current thinking is to get rid of the template matching for fids and just use circle finding.This will remove the need for setting up footprints.

markmaker, 2021-04-25:
Upsizing the template image creates a rectangular template with the corners overlapping the typical round keep-out area. With #1179 this is implicitly no longer the case, as everything is concentric. 

This does not cover non-circular fiducials, but I have never seen these on PCBs.

vonnieda, 2021-04-25:
I think #1179 does solve the problem that this was meant to address, and there's not much point in doing the specific thing in the issue, so we can close it.

closed 2021-04-25 by vonnieda