openpnp/openpnp #142: Add Importer for Altrium .pnp files

vonnieda, 2015-09-03:
See the discussion at https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/Zt92419RqKw/P-CqgGleCQAJ

Example file:

```
Designator Footprint               Mid X         Mid Y         Ref X         Ref Y         Pad X         Pad Y TB      Rotation Comment        

C1         NICHICON_A            2310mil      284.5mil       2310mil        285mil       2310mil        344mil  T         90.00 10uF           
C3         CAP0603               2130mil        325mil       2130mil        325mil       2130mil        361mil  T        270.00 1uF            
C4         CAP0603               1975mil        490mil       1975mil        490mil       1975mil        454mil  T         90.00 1uF            
C2         CAP0603               2400mil        285mil       2400mil        285mil       2400mil        321mil  T        270.00 0.1uF          
D1         SOD123                2340mil        665mil       2340mil        665mil   2412.835mil        665mil  T        180.00 1N5819HW       
L1         INDNRS5020            2340mil        515mil       2340mil        515mil       2340mil    444.134mil  T         90.00 IND NRS5020    
R2         RES0603               2050mil        325mil       2050mil        325mil       2050mil    291.535mil  T         90.00 10k OHM        
R10        RES0603               2395mil       2030mil       2395mil       2030mil   2428.465mil       2030mil  T        180.00 10 OHM         
R3         RES0603               2210mil        325mil       2210mil        325mil       2210mil    291.535mil  T         90.00 51 OHM         
U2         TSOT-23-5             2150mil        615mil       2150mil        615mil   2187.402mil    666.181mil  T        180.00 MP3302         
C5         NICHICON_A             750mil      599.5mil        750mil        600mil        750mil        659mil  T         90.00 10uF           
C7         CAP0603                650mil        600mil        650mil        600mil        650mil        636mil  T        270.00 0.1uF          
C6         NICHICON_A            1150mil      599.5mil       1150mil        600mil       1150mil        659mil  T         90.00 10uF           
C8         CAP0603               1250mil        600mil       1250mil        600mil       1250mil        636mil  T        270.00 0.1uF          
P2         52207-0460            3750mil   1965.276mil       3750mil       1920mil   3690.945mil       1920mil  T        180.00 52207-0460     
Q1         S08                   1670mil        445mil       1560mil        520mil       1560mil        520mil  T        360.00 Si4532DY       
R6         RES0603               3590mil       1050mil       3590mil       1050mil   3556.535mil       1050mil  T          0.00 10k OHM        
R7         RES0603               3590mil       1120mil       3590mil       1120mil   3556.535mil       1120mil  T          0.00 10k OHM        
R8         RES0603               3590mil       1390mil       3590mil       1390mil   3556.535mil       1390mil  T          0.00 10k OHM        
R9         RES0603               3590mil       1460mil       3590mil       1460mil   3556.535mil       1460mil  T          0.00 10k OHM        
R4         RES0603               1435mil        370mil       1435mil        370mil       1435mil    403.465mil  T        270.00 10k OHM        
R5         RES0603               1435mil        515mil       1435mil        515mil       1435mil    481.535mil  T         90.00 10k OHM        
R1         R0603             1669.999mil        280mil       1670mil        280mil   1636.535mil        280mil  T        360.00 Zero Ohm       
U3         TSSOP16_.65P_TI       3775mil       1350mil       3775mil       1350mil   3864.567mil   1460.236mil  T        270.00 TSC2046        
U1         SOT-223                950mil        600mil        950mil        600mil     1037.5mil        725mil  T        180.00 LM1117MPX      
C12        CAP0603               3600mil       1775mil       3600mil       1775mil       3636mil       1775mil  T        180.00 0.01uF         
C11        CAP0603               3600mil       1705mil       3600mil       1705mil       3636mil       1705mil  T        180.00 0.01uF         
C10        CAP0603               3600mil       1635mil       3600mil       1635mil       3636mil       1635mil  T        180.00 0.01uF         
C9         CAP0603               3600mil       1565mil       3600mil       1565mil       3636mil       1565mil  T        180.00 0.01uF         
```

cri-s, 2015-09-03:
Added pull request for named csv file as documented at:
http://www.liteplacer.com/input-file-format/

For altrium, output option need to be CSV and Metric, the above example is TxT and Imperial, and need
manual tweaking after conversion, the csv is ready to use.
Only mm is supported. Format specification must be inside the first 10 lines.

robermeyer, 2015-09-03:
Metric and/or CSV can work for Altium too.
[image: Inline image 1]
Need to document it somewhere so the Altium user knows that's the
requirement for the importer :-)

"Designator","Footprint","Mid X","Mid Y","Ref X","Ref Y","Pad X","Pad
Y","Layer","Rotation","Comment"
""
"C1","NICHICON_A","58.674mm","7.2263mm","58.674mm","7.239mm","58.674mm","8.7376mm","T","90.00","10uF"
"C3","CAP0603","54.102mm","8.255mm","54.102mm","8.255mm","54.102mm","9.1694mm","T","270.00","1uF"
"C4","CAP0603","50.165mm","12.446mm","50.165mm","12.446mm","50.165mm","11.5316mm","T","90.00","1uF"
"C2","CAP0603","60.96mm","7.239mm","60.96mm","7.239mm","60.96mm","8.1534mm","T","270.00","0.1uF"
"D1","SOD123","59.436mm","16.891mm","59.436mm","16.891mm","61.286mm","16.891mm","T","180.00","1N5819HW"
"L1","INDNRS5020","59.436mm","13.081mm","59.436mm","13.081mm","59.436mm","11.281mm","T","90.00","IND
NRS5020"
"R2","RES0603","52.07mm","8.255mm","52.07mm","8.255mm","52.07mm","7.405mm","T","90.00","10k
OHM"
"R10","RES0603","60.833mm","51.562mm","60.833mm","51.562mm","61.683mm","51.562mm","T","180.00","10
OHM"
"R3","RES0603","56.134mm","8.255mm","56.134mm","8.255mm","56.134mm","7.405mm","T","90.00","51
OHM"
"U2","TSOT-23-5","54.61mm","15.621mm","54.61mm","15.621mm","55.56mm","16.921mm","T","180.00","MP3302"
"C5","NICHICON_A","19.05mm","15.2273mm","19.05mm","15.24mm","19.05mm","16.7386mm","T","90.00","10uF"
"C7","CAP0603","16.51mm","15.24mm","16.51mm","15.24mm","16.51mm","16.1544mm","T","270.00","0.1uF"
"C6","NICHICON_A","29.21mm","15.2273mm","29.21mm","15.24mm","29.21mm","16.7386mm","T","90.00","10uF"
"C8","CAP0603","31.75mm","15.24mm","31.75mm","15.24mm","31.75mm","16.1544mm","T","270.00","0.1uF"
"P2","52207-0460","95.25mm","49.918mm","95.25mm","48.768mm","93.75mm","48.768mm","T","180.00","52207-0460"
"Q1","S08","42.418mm","11.303mm","39.624mm","13.208mm","39.624mm","13.208mm","T","360.00","Si4532DY"
"R6","RES0603","91.186mm","26.67mm","91.186mm","26.67mm","90.336mm","26.67mm","T","0.00","10k
OHM"
"R7","RES0603","91.186mm","28.448mm","91.186mm","28.448mm","90.336mm","28.448mm","T","0.00","10k
OHM"
"R8","RES0603","91.186mm","35.306mm","91.186mm","35.306mm","90.336mm","35.306mm","T","0.00","10k
OHM"
"R9","RES0603","91.186mm","37.084mm","91.186mm","37.084mm","90.336mm","37.084mm","T","0.00","10k
OHM"
"R4","RES0603","36.449mm","9.398mm","36.449mm","9.398mm","36.449mm","10.248mm","T","270.00","10k
OHM"
"R5","RES0603","36.449mm","13.081mm","36.449mm","13.081mm","36.449mm","12.231mm","T","90.00","10k
OHM"
"R1","R0603","42.418mm","7.112mm","42.418mm","7.112mm","41.568mm","7.112mm","T","360.00","Zero
Ohm"
"U3","TSSOP16_.65P_TI","95.885mm","34.29mm","95.885mm","34.29mm","98.16mm","37.09mm","T","270.00","TSC2046"
"U1","SOT-223","24.13mm","15.24mm","24.13mm","15.24mm","26.3525mm","18.415mm","T","180.00","LM1117MPX"
"C12","CAP0603","91.44mm","45.085mm","91.44mm","45.085mm","92.3544mm","45.085mm","T","180.00","0.01uF"
"C11","CAP0603","91.44mm","43.307mm","91.44mm","43.307mm","92.3544mm","43.307mm","T","180.00","0.01uF"
"C10","CAP0603","91.44mm","41.529mm","91.44mm","41.529mm","92.3544mm","41.529mm","T","180.00","0.01uF"
"C9","CAP0603","91.44mm","39.751mm","91.44mm","39.751mm","92.3544mm","39.751mm","T","180.00","0.01uF"

vonnieda, 2015-09-05:
Pull request from @cri-s merged with support for named CSV. This should handle the Altium requirement well. I have also included the above sample file in the distribution and have started a Wiki page for importing at https://github.com/openpnp/openpnp/wiki/Importing-Centroid-Data.

Please feel free to improve the page, especially for Altium as I do not own it and am just guessing at the process.

closed 2015-09-05 by vonnieda

robermeyer, 2015-09-05:
A big thank you to Jason  and cri-s for getting on the Altium.