openpnp/openpnp #148: Fiducial check (and manual check) fails with angles > 90

vonnieda, 2015-09-09:
Test case:

    @Test
    public void testCalculateAngleAndOffset2() throws Exception {
        // Create two placements to test with.
        Location idealA = new Location(LengthUnit.Millimeters, 5, 35, 0, 0);
        Location idealB = new Location(LengthUnit.Millimeters, 55, 5, 0, 0);
        
        // Rotate and translate the placements to simulate a board that has
        // been placed on the table
        double angle = 92;
        Location offset = new Location(idealA.getUnits(), 10, 10, 0, 0);
        
        Location actualA = idealA.rotateXy(angle);
        actualA = actualA.add(offset);
        Location actualB = idealB.rotateXy(angle);
        actualB = actualB.add(offset);
        
        System.out.println("idealA " + idealA);
        System.out.println("idealB " + idealB);
        System.out.println("actualA " + actualA);
        System.out.println("actualB " + actualB);
        
        Location results = Utils2D.calculateAngleAndOffset(idealA, idealB, actualA, actualB);
        
        System.out.println("results " + results);
        
        within("angle", results.getRotation(), angle, 0.001);
        within("x", results.getX(), offset.getX(), 0.01);
        within("y", results.getY(), offset.getY(), 0.01);
    }

Result is -268 (which is 360 - 92) but need to figure out why.

See thread at: https://groups.google.com/d/msgid/openpnp/afa97312-2526-40c6-9345-9bc1df71d657%40googlegroups.com

cri-s, 2015-09-09:
sorry, -268 is the same as +92 .
The error is not this, you rotate it 92 degree, check it and result is
92 degree.
One of the errors is that, the board angle is dropped and replaced
with the angle difference
of measured and ideal fiducial points.

It's personal preference if modify
 CalculateAngleAndOffset2 for subtracting rotation too, or modify the
code following the
call.

vonnieda, 2016-01-30:
This is resolved as of 5276563869193d32d305256795b9fbc481667df7. Sample job now also includes boards with > 90 and < -90 angles for verification.

closed 2016-01-30 by vonnieda