openpnp/openpnp #151: Change Icons for camera and nozzle positions

cri-s, 2015-09-12:
Red color is Alert, Stop, Break, Modify
Red color should be avoided for cruising to positions, instead it should be used to modify position using camera/nozzle position as alert color for "Attention, this changed some Values".
Just swap the icons for modify fields and for move to that position.
I personally see it more danger of unintentionally change position that cannot be restored back easily as moving the head on PNP.

vonnieda, 2015-09-13:
I disagree, but I am open to taking a vote on this. I made the movement icons red specifically because I consider the act of moving the machine more dangerous than overwriting a value. The red is there to give an indicator that you should not have your hands in the machine and that the machine may hit something when you move it.

That being said, I agree that overwriting a value can be annoying. Perhaps we can indicate it in another way? We could flash a warning that the location had been updated, and maybe allow undo?

cri-s, 2015-09-13:
The overriding value caused that two components was placed on the same
position and the resulting nozzle damage and z axis disalignment caused
machine still standing.

vonnieda, 2015-09-13:
@cri-s Do you have any suggestions on how to improve that experience? I'm open to changing the icon colors, but I think there should maybe be something more. I have overridden values several times before too.

What if we added a prompt, like: Are you sure you want to set the location to 10, 20, 30? Yes - No.

cri-s, 2015-09-13:
>
> @cri-s Do you have any suggestions on how to improve that experience? I'm
open to changing the icon colors, but I think there should maybe be
something more. I have overridden values several times before too.
>
> What if we added a prompt, like: Are you sure you want to set the
location to 10, 20, 30? Yes - No.

If over 300 position need to be corrected, then the security question is
real time killer.

I don't know how hard is adding undo for location value.
If after change the background goes red and undo exist, I think it would
resolve the problem.

Problem is only boardlocation and placementlocation.

cri-s, 2016-08-23:
Solution is to remove the button and use key modifier
That change icons and check it inside action.
If Ctrl is pressed, icon change color and have store functionality.

cri-s, 2016-08-23:
This issue is obsolete because of read only functionality that will be merged shortly.

closed 2016-08-23 by cri-s