openpnp/openpnp #153: config file xml

cri-s, 2015-09-14:
When changing api and attribute or element get removed, reading old files
Causes exception for attribute or element not known.
Unknown attributes should not raise up exceptions.

vonnieda, 2015-09-14:
This is not possible with the current XML serialization framework due to a bug in the framework. The framework does support adding a version field but there is a bug that causes it to fail for subclasses.

I suggest backing up your files before switching major API versions.

cri-s, 2015-09-14:
Then maybe it is better to use json files for 
Job files.
Actually its not a issue, but on production yes. If doing board and upgrading sw, the old files are anymore producable and for creating new files, the work of associating packages,  checking data and adjusting it is really ass n paid redoing it.

vonnieda, 2015-09-15:
I agree, I want to switch to JSON someday, but it's a huge change for little value.

All of your files should be forward compatible, i.e. if you have an old file and new software it should load fine. If not, please let me know which versions break it.

vonnieda, 2015-10-04:
I'm closing this as won't fix. Files are forward compatible and I don't think it's worth the effort to try to be able to produce old versions of files.

closed 2015-10-04 by vonnieda

cri-s, 2015-10-04:
I agree, depending of target.
For hobbyist it is no issue. For professional, if after 1 year producing
new batch and the newer version don't load file this can be problematic.
If knowing that, software version need to be backup and changed
configuration need to be backported using macro processor for
semiprofessional use.
Please think about including board and job creation version sw on config
files.
Creation version, not changed version.

vonnieda, 2015-10-04:
@cri-s New versions should always be able to load older files. It's only the opposite that is a problem.

For instance, if you create a file today and then try to load it on newer software 1 year from now, that should work. But if you create a file today and then try to load it with software that is 1 year old, that is not guaranteed to work.

cri-s, 2015-10-04:
Simple example:
Bottom assembly actually don't work or don't completly work.
For implementing it, it is necessary to get the board dimension on X
and Y and i'm sure that
this field is required, not optional, because it's needed for other
purposes, at example fiducials and maybe safety checks for feeder
positions.
Ok, you could change the Head offsets, but this is improbable.

If you load the 1 year old file having only top assembly, then
simpleframwork.XML comes out with
missing value, required=true ... and it's not possible to load that
file, because in the meantime
it happened that for previously not knows reasons some values are
additional needed.

my 2cents