openpnp/openpnp #154: getSelectedCamera() sometimes returns a fixed camera, causing movement errors.

vonnieda, 2015-09-15:
Many places in the code use getSelectedCamera() to determine which camera should be used for an operation, e.g. moveCameraToLocation which is used all over the place.

If the selected camera is a fixed camera, this results in a movement error because fixed cameras can't be moved.

We need a way to readily find the "right" camera to move when a different camera is being viewed.

One way to solve this is to define a set of functions that a camera can perform. Examples would be: Fiducial Vision, Bottom Vision, Table View, Positioning. Some cameras might have more than one of these functions defined. For example, a standard head mounted camera will probably have Fiducial Vision and Positioning defined.

With this in place the API could be changed to something simpler like getFiducialCamera() or getCameraForFunction(Fiducial) which leaves no room for ambiguity.

cri-s, 2015-09-15:
> Closed #154 <https://github.com/openpnp/openpnp/issues/154> via 1c8db2d

Generally there is one downlooking camera and several uplooking cameras.
But if two independents head are used two down looking cameras can be used.
Generally cameras are looked to nozzles were different heads have different
nozzle numbers for the same nozzle, here it works with hierarchy. As
example it is common to have tree up looking cameras. As example one with
5cm width, another with 3cm and the third having 1cm field of view. It is
the nozzle that indicate what component are to inspect with with camera and
for middle size nozzle two cameras could be used and selected probably
based on distance.
Further this resolve the issue that only the down looking camera attached
to the same head is used. Any other strategy usually gives problems.

vonnieda, 2015-09-16:
Thanks Cri, this is good information!

vonnieda, 2015-09-17:
Reopening this as I've been thinking about it more and hitting the problem more often. Started working on a temporary solution:
Machine.getDefaultHead()
Head.getDefaultCamera()

General idea is that the camera selection panel is no longer queried for the selected camera but instead f we use the currently selected nozzle (or paste dispenser, in the future) like this:

MainFrame.machineControlsPanel.getSelectedNozzle().getHead().getDefaultCamera().

My thinking is that the camera panel becomes "view only" and we don't really care which camera is selected, we just reference the camera that makes sense for the tool you are working with.

Still refining this idea, just putting my thoughts here.

cri-s, 2015-09-17:
Why not
NozzleTip.getCamera()
And
Base.getCamera(PlacementSolution.getNozzleTip())

PS.
Base is a head that is present for actors and cameras or other  controls
that are attachable to heads and need to be  placed on table instead of
real heads where the interfaces remains the same. This allows having as
example light controls driven by pwm outputs of motion controller without
the need to add lightning class. Same is true for dispenser and flux
station or flip station.
It have id 1 and head 0 is only proxy for real heads that begins with 2.
Head 0 is the last head moved, or if null, head nr 2.
It should be different, but for historical reason the base cannot be 0.
Maybe one can add isTable  and getTable but actually its hardcoded with the
IDs.

vonnieda, 2015-09-18:
I will think more about the NozzleTip.getCamera() suggestion and get back to you.

As for Base, you should use Machine for this. It already has a Camera list for fixed cameras, and in your Machine implementation you can easily add any other objects you want such as Actuators.

cri-s, 2015-09-18:
Having base as head have advantages for camera code.
For centering on nozzle it can use the same code as for centering on home
position.
It works because base returns NdN as location. Same is true if camera need
rotation for closed loop vision. The same is true for code checking driver
movement driving 6mm away and measures distance using camera. Because up
looking camera have narrow field of view the value is genrrally more
precise.
Temperature on ALU can give misalignment if not corrected
And if not doing fiducial lookup the pick position of feeders are usually
compromised.

commit 1c8db2da (2015-09-15): Change the order that cameras are added during bootup so that the first head camera ends up selected instead of the first fixed camera. This is a temporary hack until we fix #154.
commit e33bcb7e (2016-01-01): Final bits of refactoring and cleanup. Fixes #154.
closed 2016-01-01 by vonnieda