openpnp/openpnp #160: question

cri-s, 2015-10-07:
Why the sequence number gives problems? It works after some time, but initially gives problems.

      public Board() {
                this(null);
                this.name= "Board"+ Configuration.get().getMachine().getJobProcessor().getJob().getNr();
        }

getJobProcessor() is equivalent to getJobProcessors().get(JobProcessor.Type.PickAndPlace)

        public int getNr() {
                return ++nr;
        }

vonnieda, 2015-10-07:
I don't understand what you are asking. Can you detail the problem? This getNr() code is not in OpenPnP.

cri-s, 2015-10-07:
Yes, getNr is not part, and for this i have given the definition of
the function.
It was added to give automatically "Board1" "Board2" names when
teaching and not loading
centeroid positions instead of the question box about Names. Same with
Placement ID.
At the beginning, it don't work, and gives some problem when loading job/boards
(null pointer exception).
After some time (minutes), it works without problems.
It is not traceable for me what is the problem,
maybe some listener. If you have idea why this could gives problems, i
would know it, otherwise
close the question as investigating this problem is hard and it's not
worth the time.

vonnieda, 2015-10-07:
When the Board constructor is called the Configuration is not yet loaded, so Configuration.get().getMachine().getJobProcessor() will return null. If you want to assign the sequence when creating a new Board you should set the name externally, when you create the Board. If you want to assign the sequence every time a Board is created (seems unlikely) then use a ConfigurationListener. Something like:

        Configuration.get().addListener(new ConfigurationListener.Adapter() {
            @Override
            public void configurationComplete(Configuration configuration)
                    throws Exception {
                // assign name here
            }
        });

closed 2015-10-07 by vonnieda

cri-s, 2015-10-07:
It works now.
One last question that is open.
Actually it don't work from begin, it works only after switching from
solder-paste to pick-and-place tab.
Then it works adding new board indipendently from board name question.
Why and is this fixable ?

2015-10-07 6:16 GMT, Cri S <phone.cri@gmail.com>:

vonnieda, 2015-10-07:
@cri-s This is due to how jobs are loaded into the job processor. The job is only loaded when the tab becomes active. This is not an ideal design, and I have a note in the code to correct it, but I have not had time yet. For the time being, unless you are using solder paste I recommend just removing the solder paste job processor from your machine.xml and you will avoid the problem.