openpnp/openpnp #164: placements panels

cri-s, 2015-10-13:
Placement panel should be the first
Panels. This should resolves the issue of not adding or loading boards/job if the user have not clicked on place tab.

vonnieda, 2015-10-13:
I considered this, but I felt that it made more sense for it to be the second panel because in a given job you would typically do solder paste before placement.

What do you mean about "This should resolves the issue of not adding or loading boards/job if the user have not clicked on place tab."?

cri-s, 2015-10-13:
If having stencil because of free stencil on board order or cheap laser
plastic foils or made with cameo or for timing reasons
Requirement can be different.
Dispensing is more glue points and additional solder paste for components
needing more solder as stencil gives.

Until the placement panel is clicked it's
Not possible to add new board or load existing board. I have already asked
you and the response was that actually its so because class is initialized
only after panel is build/loaded.

FlyingLotus1983, 2015-10-13:
I also find it a bit weird that it always goes to solder paste, however I think an easier approach would be to just have this a configurable attribute that stored the last panel state, which would be recalled whenever the UI started. This seems the most logical because you can't predict what a user wants to do with the machine, but it is easy to remember and recall the last thing that was done.

vonnieda, 2015-10-13:
My reasoning is this:

* If the user doesn't have a solder paste dispenser the panel is never even
shown. It defaults to Placements.
* if the user does have a solder paste dispenser then that is probably what
they want to use first in a given job.

I'm fine with changing the behavior to show Placements first, if it really
bothers you. A preference would be fine too.

As for "Until the placement panel is clicked it's Not possible to add new
board or load existing board.", this is simply not true. You may have
misinterpreted my comment on the JobProcessor stuff. Can you explain or
show (in a video) exactly what you are doing and what is happening? You may
just be hitting a bug.

Jason

FlyingLotus1983, 2015-10-13:
The solder paste automatically shows up when using the default XML, so some users might not even know that the tab can be disabled.  This issue might be moot after the GUI is able to allow the paste job processor to be removed, but at the moment there's no way to get rid of it other than manually removing it in the XML, if a user didn't need the functionality.

cri-s, 2015-10-13:
If when openpnp starts I make new job and then add new board or load a
board no line is inserted to the job table. When changing to place panel
instead of solder dispense panel,
The buttons or menu entry now works.
I have accounted it as know bug but i loosed some time to find out how it
works. Previosly I have
Removed the name dialog as I prefer having automatic naming generation and
if needs change it on table later. This change surely don't change the
behavior. If needed I can make video.

vonnieda, 2015-10-13:
This issue is not in the develop branch. Perhaps your changes are causing
the problem.

Demo video here: https://www.youtube.com/watch?v=kpqU63ebgTw

Jason

cri-s, 2015-10-13:
Then it must be Linux specific issue or my used java libs.
I can reproduce the problem using unmodified version too, it quits with error
"unable to save" and one click to job panel resolve the problem. Then
with same filename
and same dir it works without problems.

Beginning of the logs indicating some basic infos:

The output of logs:16:21:01,143 INFO                  Main Launching
the application...
16:21:01,171 DEBUG                 Main Operating System: Linux
3.13.0-63-generic
16:21:01,173 DEBUG                 Main JRE: 1.7.0_80
16:21:01,174 INFO                  Main Java Launched From:
/usr/lib/jvm/java-7-oracle/jre
16:21:01,174 DEBUG                 Main Class Path:
/home/av/rep/openPnP-dev/o/openpnp/target/openpnp-gui-0.0.1-alpha-SNAPSHOT.jar
16:21:01,175 DEBUG                 Main Library Path:
/usr/lib/jni::/usr/java/packages/lib/i386:/lib:/usr/lib
16:21:01,175 DEBUG                 Main User Home Directory: /home/av
16:21:01,176 DEBUG                 Main User Working Directory:
/home/av/rep/openPnP-dev/o/openpnp
16:21:01,199 DEBUG                 Main OpenPnP 0.0.1-alpha-SNAPSHOT
build fe2538 Started.

code for this output if you are interested:
    logger.info("Launching the application...");
    logger.debug(
      "Operating System: " + System.getProperty("os.name") + " " +
      System.getProperty("os.version")
    );
    logger.debug("JRE: " + System.getProperty("java.version"));
    logger.info("Java Launched From: " + System.getProperty("java.home"));
    logger.debug("Class Path: " + System.getProperty("java.class.path"));
    logger.debug("Library Path: " + System.getProperty("java.library.path"));
    logger.debug("User Home Directory: " + System.getProperty("user.home"));
    logger.debug("User Working Directory: " + System.getProperty("user.dir"));

vonnieda, 2015-10-13:
Could be related to Swing event ordering, which is sometimes different on
different platforms. I'll check it out.

Jason

vonnieda, 2016-01-13:
I've moved the solder paste panel to be second to avoid confusion. The original event ordering issue still needs to be fixed. I've made two attempts to fix it but both turned into large refactors and I reverted them. The entire UI for placement/paste needs to be reconsidered so for now I'm just disabling it by default. To re-enable it see https://github.com/openpnp/openpnp/wiki/Paste-Dispensing.

closed 2016-08-23 by cri-s