openpnp/openpnp #166: Revisit Board Locations

vonnieda, 2015-10-19:
It seems there are some serious problems with the design of Board Locations. Panels are not well supported, centers are preferred by many and panelization is hard using a 0,0 origin.

This has been discussed in several other issues but this one is to collect them all and provide a central discussion point for a possible design modification of board locations.

Supporting references:
https://groups.google.com/forum/#!topic/firepick/1Fnr80QWnqA
#149 
#128 

I'd appreciate if interested parties ( @cri-s , @sukarsa, @FlyingLotus1983 ) would comment on this issue as to how they would change board locations, keeping the following in mind:

* Support panelized boards.
* Don't change the board location on fiducial update. (Bias point?)
* Don't require the user to put the board at machine 0,0. 
* A job can contain multiple different board designs and these boards may appear anywhere on the machine bed.
* Bottom coordinates are logical.
* Data can be imported directly from common CAD packages without manual intervention.

I am particularly interested in how to handle bottom coordinates when you consider how these are typically exported from CAD packages.

cri-s, 2015-10-19:
Board need width and height  information too.

Width is used for bottom part placement and both is needed for fiducials
validity.

The need to have machine zero equal to PCB zero is derived from the fact
that openpnp don't have PCB zero location as usual for PnP and this was the
easyest workaround for that

Real X coordinate is: width-X for bottom parts.

vonnieda, 2015-10-20:
@cri-s Can you provide details on what fields or variables you'd like to see for setting up a board and job? Doesn't have to be within the confines of how OpenPnP does it now. If you were writing OpenPnP for the first time how would you handle board locations?

I'm particularly interested in what you mean by "openpnp don't have PCB zero location as usual for PnP". 

If you could just list out the fields that you would use that would help a lot. For instance, right now a Job contains a list of BoardLocations and a BoardLocation contains a machine Location that says where on the machine to find a board's 0,0. How would you change this?

cri-s, 2015-10-20:
Every pnp have one internal config parameter that assigns the location of
the PCB. Openpnp don't have it explitly, however setting up head and camera
offsets correctly it is possible to have coordinate 0:0 for PCB location.
This is where user place pcb usually in order to usually not need specify
offset for panel or single board. Using head offset work good and have
further the advantage to have same pos between openpnp and other external
tools for components.

Boards need height, width and thickness info
Machine need flag if using thickness info, if board alignment is left or
right based and if it should cycle automatically or manually.

Board alignment is right edge based if it's loaded Using belt conveyor
because then rc servo actuate board stopper..

Thickness depends on the machine construction if it need to be considered
or not. If PCB clamp pushes up, no height adjustment is necessary. Further
this have the advantages of having the possibility to use inexpensive
support pins.
If clamp push down height need to be considered. Disadvantage of clamping
up is the reduced height clearance on bottom.

Height is used only for fiducial inside panels.
Otherwise it don't matters.
Width is needed for calc the coordinates on top or bottom side depending on
the alignment flag. Depending on flag, if coordinates are negative or
positive error need to be raised.
For left alignment. Bottom x coordinate is board_width - X .

vonnieda, 2015-10-20:
@cri-s I think I understand how you are using offsets and board locations now. Tell me if this is right:

* You set up your nozzle and camera offsets so that if you command them to move to 0,0 they land at the place where you normally put your PCBs.
* When you create a job you just leave board location set to 0,0 since this is the same place you put all PCBs.

Further, I now understand your process for board bottoms. Instead of moving the board location you leave the board location at the same place (0,0) and just do width - x when placing on the bottom. 

Your setup is designed around the concept of there being one corner that you always place boards at. I see that this would be useful and I realize that many pick and place softwares do it this way. I am not sure that I agree that this is how OpenPnP should do it, but now that I understand it I will think about it. Thanks for clarifying.

cri-s, 2015-10-20:
Correct. Maybe I should send you two examples of panels and you tray to
setup and assemble both top and bottom, print out the image for setting up
the job.
Tell me if you want receive production data.

vonnieda, 2015-10-20:
Yes, please do. If it's public data please attach it to this issue. If it's
private, please send to me personally at jason@vonnieda.org.

Thanks,
Jason

cri-s, 2015-10-20:
It is possible to use carrier and assemble more boards spacing them over
carrier.
Then its the same as programming panels
Except board positions is different for top and bot If not using flip board
strategy.

vonnieda, 2015-12-15:
Note to self: See email from phone.cri@gmail.com on Oct 20, 2015 for panel files.

vonnieda, 2016-01-30:
A recent PR from @cri-s added board dimensions and changed how bottom locations are handled. I believe this solves the immediate problem so I am closing this. We already have another active ticket for adding panel support, so I think that will suffice to cover the rest of this one.

That ticket is #128

closed 2016-01-30 by vonnieda