openpnp/openpnp #167: Clicking stop after unplugging smoothie board locks up UI

embmicro, 2015-10-19:
If you click start, unplug the smoothie board, then click stop, the UI locks up. I just lost a bunch of project settings because of this (teach me to save more) :/.

vonnieda, 2015-10-20:
Are you using the OpenBuilds driver?

If so, I see the bug. It should be set to timeout after a period and it's
not.

Ideally the UI needs to be decoupled from this anyway. That's how I will
plan to fix this.

vonnieda, 2015-10-20:
Note to myself: Easy fix will be to run setEnabled() on the machine executor.

embmicro, 2015-10-20:
I am running it on Linux and just running everything from the tar archive.

vonnieda, 2015-10-21:
Bug should be fixed now if you download a fresh copy.

vonnieda, 2016-02-13:
Reopening because while the UI no longer locks up, the setEnabled(false) still doesn't time out which makes it impossible to actually close the program. Preparing a fix for that now.

commit 28487723 (2015-10-20): setEnabled now runs on the machine executor which fixes #167
commit a5500756 (2016-02-13): Fixes #167 (again) by enabling timeouts in the OpenBuilds driver. Also converted the driver to the newer LinkedBlockingQueue model.
Changed default timeout for commands in the OpenBuilds driver to 5 seconds instead of forever.
Changed home to allow 60 seconds to timeout because of the above.
closed 2016-02-13 by vonnieda