openpnp/openpnp #172: Machine Setup Tab -> ReferenceMachine -> Heads -> ReferenceHead H1 -> Cameras -> Panel

JacobChrist, 2015-11-16:
On the Machine Setup Tab in the ReferenceMachine tree under the Heads -> ReferenceHead H1 -> Cameras, if you select a camera the panel to the right has two tabs.  The tab text is in correct.

The tab heading when selecting a camera read "Untitled" and "Untitled".  They should probably read "General Configuration" and "Camera Specific".

JacobChrist, 2015-11-16:
This is coming from approximately line 169 of this file:
openpnp/src/main/java/org/openpnp/gui/MachineSetupPanel.java

It also affect every panel not just cameras.

vonnieda, 2015-11-16:
Thanks for the bug report. I'm aware of this and it's something on my list
to fix, but it's good to have an issue to track it anyway.

JacobChrist, 2015-11-16:
What's your plan on fixing this?  I'm trying to fix it (mostly to understand the code base not because I think it needs to be fix).  If I can understand how you might go about fixing this this I might be able to patch.  From my preliminary looking, it seems as if you need to have each panel implement the PropertySheet class so that the title is accessible when rendering the control.  I though this would be a simple fix three hours ago.

vonnieda, 2015-11-17:
Yep, that was the basic idea but I have had a few false starts on this. I
actually think this interface is going to get redesigned before this gets
fixed because as it is right now there are some interlocking dependencies
that I hate. So, as of now I don't really have a plan. It's part of the
much bigger task of finishing up the Machine Setup tab which has been in
progress for quite some time.

vonnieda, 2015-11-17:
See #103

dave-prosee, 2016-11-05:
@vonnieda : **you are probably aware of this, but I use this as a way to get familiar with the code and appreciate some feedback a.k.a. insight.** 
The configuration wizards for Jpanels don't set the tab title. Tab title is set by e.g. their grandparents, i.e. the CamerasPanel, FeedersPanel or MachineSetupPanel. The first two have code with specific text to do that. The MachineSetupPanel tries to get the title based on the object (type) but that info is not available. Could it be a solution to add a <confWizardTitles> attribute to the machine.xml. At least for now. So we can access titles via the configuration class?

dave-prosee, 2016-11-05:
Or, like the Gcodedriver, add the title in the getPropertySheets method like for OpenCvCamera:
 @Override
    public PropertySheet[] getPropertySheets() {
        return new PropertySheet[] {
                new PropertySheetWizardAdapter(new CameraConfigurationWizard(this), "General Configuration"),
                new PropertySheetWizardAdapter(getConfigurationWizard(), "Camera Specific")};
    }

vonnieda, 2016-11-05:
Hi @dave-prosee, yes, your second point is the way to go, as GcodeDriver does it. I haven't spent much time messing with this because, as noted above, I'm not very happy with this entire API. It was a first stab at a new idea that never quite got finished. I would definitely support getting as many of the titles in as we can, though. It's sloppy to have a bunch of tabs called "Untitled".

dave-prosee, 2016-11-05:
OK if I toss in a few of those changes? Separatly?

vonnieda, 2016-11-05:
Absolutely! Thank you!

dave-prosee, 2016-11-05:
Hello @vonnieda , I made two PR's. One for camera's and one for feeders. The name for the titles were taken from the Feeders and Cameras panels. Driver(s) was/were already OK. For the remaining objects I would like to know what you want for titles. So please specific titles for:
- Signalers (btw do not have experience with that)
- Heads and nozzles
- Actuators
- Job processors (i have now only the PnPjobprocessor and remember there is also a pastejobprocesser)
- Vision (bottom vision with enabled and pipeline, fiducial locator).

Thanks.

vonnieda, 2016-11-05:
@dave-prosee I think that we can just use Configuration or General Configuration (dealers choice) in all cases where only one property sheet is being returned. In the case of multiple, which is rare, we can use the above for the default sheet and then I think the additional ones should be self evident. I just took a quick look through and I don't think there are actually any with multiple sheets aside from the cameras which you've just fixed.

Also, I like your idea of "if title = null then title = "Configuration". This is a one line change at https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/gui/MachineSetupPanel.java#L161

Jason

dave-prosee, 2016-11-06:
Done. See the upcoming PR.
I have removed the getPropertySheets method override from the LtiCivilCamera so it uses the base class one and the code is neater.
There are three types of objects that return titles now (feeders, cameras, gcodedriver) and two of them have two specific tabs (cameras and gcodedriver).
Reading back what the issue is about, I think this issue could be closed. It was good as a starting point and I am open for a more difficult one.

vonnieda, 2016-11-06:
PR merged, thanks @dave-prosee!

closed 2016-11-06 by vonnieda