openpnp/openpnp #176: Ability to zero out nozzle rotation

readerror67, 2015-12-09:
Depending on the offsets generated during a fiducial check, its possible for the nozzle to be rotated at a value which cant be manually zeroed back out (movement resolution 0.01mm but rotation may be 0.123). It may be useful to have a command to re-position them back to their home(0) position. Mostly noticed when setting locations based on camera/nozzle position.

vonnieda, 2015-12-10:
Related to #175, not sure what the right solution is yet. I've seen this issue as well, although I might argue that it doesn't matter since we seldom capture nozzle rotation. 

Some possible options:
1. Add a zero axis button to the jog controls.
2. Add a zero axis button for each axis.
3. Add a "home this axis only" button for each axis, similar to Pronterface and others.

cri-s, 2016-01-11:
You should use the existing "move to 0" as it remove any offset as long this command exist.

vonnieda, 2016-02-13:
Revamped the jog controls panel to include pretty much all of the above. We can't do "home this axis only" due to the driver model, and I don't think it's worth changing that much code to add it, but I did add "zero this axis" buttons

New jog panel looks like:
[image]

closed 2016-02-13 by vonnieda