openpnp/openpnp #177: ReferenceStripFeeder Error -- Data collection

readerror67, 2015-12-09:
Ended up putting some breakpoints in place if the error took place, ended up collecting a screenshot and the debug values from Eclipse:

[image]
(Sorry about the picture, pasting seemed to really mess with Githubs formatting and it was unreadable) 

Screenshot:
[image]

readerror67, 2015-12-09:
Adding pictures with light on and off.

On:
[image]

Off:
[image]

vonnieda, 2015-12-10:
Note to myself so it doesn't get forgotten: This data was collected in trying to track down why ReferenceStripFeeder vision was failing sometimes. I noted the pictures look somewhat out of focus. @readerror67 followed up saying that eventually the expected position was getting so far out of whack due to the line being off at a distance that it wasn't finding the hole.

One solution is to recalculate the line for each feed by using the last found vision hole and either the first, second or next more recent vision location.

vonnieda, 2015-12-14:
This is now fixed, or at least improved. The vision code now uses the last located hole along with the first hole to calculate the line to follow and it results in much  better following at the extents of strips.

Fixed in 7ee64b0624ec89ad5bd13fe3e5e2c54f713d5191

closed 2015-12-14 by vonnieda