openpnp/openpnp #178: strip feeder error when his angle is 180

JoseDanielMunozFrias, 2015-12-15:
When a strip header is located at 180 degrees, the software thinks that it is at 0 degrees and go to the wrong side to pick the part.

vonnieda, 2015-12-15:
Hi @JoseDanielMunozFrias, can you post a screenshot or some details to show how to reproduce this problem?

JoseDanielMunozFrias, 2015-12-15:
Whow! You are pretty fast. I have sent a commit with the solution.

JoseDanielMunozFrias, 2015-12-15:
Whell, A pull request.

JoseDanielMunozFrias, 2015-12-15:
[image]

vonnieda, 2015-12-15:
Thanks @JoseDanielMunozFrias, I will test this and merge if it looks okay in a day or two.

vonnieda, 2015-12-16:
Thanks for the report and the pull request!

closed 2015-12-16 by vonnieda

JoseDanielMunozFrias, 2015-12-16:
Thanks to you for this phenomenal project.