openpnp/openpnp #180: Wiki Images - Ignore

vonnieda, 2015-12-18:
Placeholder for Wiki Images. Not a real issue.

vonnieda, 2015-12-18:
[image]

closed 2015-12-18 by vonnieda

vonnieda, 2016-01-22:
[image]

vonnieda, 2016-01-27:
[image]

vonnieda, 2016-02-21:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-06-18:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

vonnieda, 2016-11-04:
[image]

JacobChrist, 2016-11-04:
Jason,

Correct me if I'm wrong. Your posting these images so that you can link to
them in the Wiki documentation.  Might it be a good idea to store them in a
directory in the repo?

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

vonnieda, 2016-11-04:
> +1 (909) 652-0670 <(909)%20652-0670> Phone

Hi Jacob,

You've got it right. I use this issue to store images that I am using in
the Wiki. I do it this way because Github supports image drag and drop into
Issues but not Wiki pages. So, I drag the images here and copy the URL that
gets generated for the Wiki. I find putting them in the repo to be a pain
and it really slows down my workflow for creating Wiki pages.

Jason

JacobChrist, 2016-11-04:
> a289-11e6-8350-298314ca5e2f.png
> 2WAVpVFDhKHHgQ0xRNMhE8ks5q62q3gaJpZM4G336-

Okay... After I sent the e-mail I thought it might make the repo bloated as
well.

Jacob

Jacob Christ
*ProLinear/PONTECH, Inc.*
+1 (909) 652-0670 Phone
http://www.pontech.com

vonnieda, 2019-06-13:
[image]

markmaker, 2022-08-10:
[image]
[image]