openpnp/openpnp #183: Refactor NozzleTips to be global

vonnieda, 2016-01-03:
Currently you have to duplicate nozzle tips for nozzles as this implies that in a multi tip system you have a bank of tips for each nozzle. I think this is unlikely and makes the system inflexible. This should be changed so that nozzle tips are a machine level object (instead of nozzle) and we just mark which ones are compatible with which nozzles. This makes it possible to load any compatible nozzle tip into any nozzle without having to have duplicate tips.

This will also make it easier to add better support for specifying packages compatible with nozzle tips without repeating data, and the same for vision information.

cri-s, 2016-01-11:
There is only some remarks to be considered:
It can be that nozzles have fixed nozzle-tip.
As example two nozzle machine where one nozzle is fixed and the other can be changed.
Reason for doing that, non changeable nozzleTips have higher precision that automatic
changeables nozzles.
Some nozzleTips are compatible only with certain nozzle. 
Example are higher vacuum needed nozzles that can be used only on one specific nozzle
because machine construction gives that nozzle additional vacuum.

Annother consideration is NozzleChanger.
It's handled on machine specific implementation (machine/reference)
This is bad, because JobPlanner don't know how many Nozzles can be changed during Job processing as the api don't allow it.

vonnieda, 2016-01-11:
Thanks @cri-s, these are all good points. Please see my comments on #188 for how to get access to all the information about NozzleTips.

As part of the refactoring for this ticket I will make a new NozzleTipChanger interface that can be extended for different types of changers: manual, 3 position, carousel, etc.

cri-s, 2016-01-11:
Jason, i have extended the driver api a bit,
void dwell()  , wait until finished
void dwell(double)  wait time in seconds.
boolean GCode(string)
boolean GCode(string,time)

Actually the only not right is on firepick driver, where i probably
need community help if this
get accepted by you. Then i suggest to use GCode for driving it. I
suggest then to use GCode
for driving changer and i think, manual should be highest priority.
Currently i have tested only dwell(double) in order to go away with
thread sleeping and that
works.

vonnieda, 2016-01-11:
void dwell()  , wait until finished
void dwell(double)  wait time in seconds.

What is the purpose of the dwell commands and how will they be used? I don't see this as being included because all of the driver methods are supposed to block until they are finished. If something is returning before it's complete then that is a bug.

boolean GCode(string)
boolean GCode(string,time)

These we will not include. The Driver interface is designed to be controller agnostic and this would make it specific to GCode. If you want GCode specific methods just extend the ReferenceDriver interface and implement them in your branch.

cri-s, 2016-01-11:
dwell wait until all gcode are processed. No only some gcode are
blocking, the other return without being completed.
dwell(double seconds) wait seconds until and return after all previous
motion code are finished.
I have used it to replace    Thread.sleep(pickDwellMilliseconds);  as
otherwise it don't worked
because gcode and pc are never in sync.
GCode() would be nice for nozzle changer code and others, that are
always machine specific
and mostly have only moves inside it. You'r decision.

vonnieda, 2016-01-11:
@cri-s All driver calls should block until they are finished. This contract must be enforced or lots of other things will break. For instance, if moveTo() or place() does not block then the machine may try to move before it's finished placing. If you have driver calls that don't block, that is a bug and should be fixed. Just call dwell() internally in the driver, as many of the drivers already do.

We can't add GCode specific things to the driver interface because there are machines that don't use GCode. For instance, the TinyG and FirePick drivers use JSON for communication. Having a GCode command on it would be meaningless.

If it's very important for you to be able to send generic commands to the driver I would be okay with adding something like `public Object sendCustomCommand(Object);`. With this you could do something like `sendCustomCommand("M100") and then process that as Gcode within your driver.

That being said, I suggest you extend ReferenceDriver and add these functions to your own driver. The program is designed to be extended like this.

cri-s, 2016-01-11:
moveTo don't need to block, and it's contra productive. It's enought
that after pick and place
the call is blocking, and as example actors are not blocking too. .
GCode works for TinyG , FirePick , don't know i have left it as dummy
function for now.

vonnieda, 2016-01-11:
@cri-s If you want to continue this discussion, please move it to another issue or thread. This issue is for a refactoring of NozzleTips, not Driver.

vonnieda, 2016-01-18:
Also see #89.

markmaker, 2018-10-09:
Hi

This revision might be the chance for some simplifications. 

IMHO the nozzle tip compatibility mapping UI should be on the package, not on the nozzle tip. This way the list has a more or less fixed small size i.e. is more manageable for large package data bases. If you add a new package you can quickly choose the right nozzle tips right there.

**Or Even Better**

Simpler but more effective, one could instead specify a compatible nozzle tip **suction diameter** range instead of selecting specific nozzle tips. The nozzle tips themselves then just have to specify their **effective suction diameter** (more about that below).  

This way, if you later add a new or duplicate nozzle tip, it is automatically matched up to all existing packages. 

This also opens up for a (future) automatic match-up based on package Body Width/Body Length (whichever is smaller) that in turn may be imported from the ECAD. 

Furthermore the package data would become portable across machines so the OpenPNP community could better share it. 

As already explained, each nozzle tip just needs to specify its **effective suction inner diameter** to be matched against packages. Users could slightly adjust the **effective** value to be different from the actual physical diameter to allow for differences between tips/vacuum strengths and other preferences. 

Ideally (and for migrating existing setups) the two systems could be combined. The manual section of the tip(s) overriding the **suction diameter** match-up.

_Mark

vonnieda, 2018-10-09:
Hi @markmaker,

I agree on point one. Management should be done in the packages UI. The reason it was done the way it is was for expandability. As it is now, you can completely replace the nozzle tip change system with your own if you have some crazy kind of system. Ultimately, I don't think it's important for it to be this flexible, so making it a high level item works.

The second point, about using suction diameter, won't work in general practice. There are a lot of nozzle tips that are specific to a package and the suction isn't really related. For instance, there are tips that have a rounded cup for MELF diodes, and tips that actually have vacuum actuated mechanical grabbers for oddly shaped parts. A quick browse through http://www.firepick.org/assets/pdf/juki-nozzle-catalogue-rev-c3.pdf will show what I mean.

While those may be rare in practice, I don't see any reason to limit people from using them.

In general, people won't change their list of nozzle tips often, and once they have built a package library that won't change often, either. So I think assigning nozzletips to packages is good enough.

markmaker, 2019-05-31:
Hi Jason

seing [in the group ](https://groups.google.com/d/msg/openpnp/7SckWf-sWkE/O1cmRLFfAwAJ )that you are working on this, I wanted to reiterate point one of [my earlier post](https://github.com/openpnp/openpnp/issues/183#issuecomment-428094537).

> IMHO the nozzle tip compatibility mapping UI should be on the package, not on the nozzle tip. This way the list has a more or less fixed small size i.e. is more manageable for large package data bases. If you add a new package you can quickly choose the right nozzle tips right there.

As you already need to migrate (or break) existing configuration in a big way, this might be right moment.

If I'm kicking at open doors, all the better :-)

_Mark

vonnieda, 2019-05-31:
Hey @markmaker -  yep, I agree. I think it belongs on Package as a default, and maybe also Part and Placement as an override. But I'll start with Package.

vonnieda, 2019-06-01:
@markmaker or others: Any thoughts on how we should specify the package -> nozzletip link? I'm thinking about adding a type field to nozzletip. So a typical setup might have the following nozzletips:

|  Name | Type  |
|---|---|
|  500-1 | 500  |
|  500-2 |  500 |
|  505 | 505  |

Then package would reference by type, rather than name. Name would be used by the user to help them identify a specific nozzletip, while type would be used to identify any one of a number of compatible nozzletips.

Additionally, do you think the package -> nozzletip link needs to be one to many? I'm thinking one to one for now.

markmaker, 2019-06-01:
Hi @vonnieda 

> I'm thinking about adding a type field to nozzletip. 
>package would reference by type, rather than name

Abstraction from a specific nozzle tip is a good idea. LED matrix assemblers with many times the same tip would be happy.

> needs to be one to many? I'm thinking one to one for now.

I see it would remove complexity. But...

I'm no expert but looking at typical specifications for nozzle tips, there are usually multiple and overlapping package sizes specified for each tip size. If you sort the parts and tips by size, the relationship is like a "sliding window" both ways.  

Some would want to use the nozzle tips more flexibly in order to reduces nozzle tip changes, saving job time. 

Perhaps it would be an option to go for a nozzle tip **type range**, instead of a single type. But I doubt it would be less complex than a multi-selection list. And there might be odd cases where you are glad to be able to pick compatible tip types as you like. 

_Mark

vonnieda, 2019-06-01:
Agreed - thanks Mark.

Jason

closed 2019-06-11 by vonnieda