openpnp/openpnp #188: Job Planning - access to machine information missing

cri-s, 2016-01-11:
Job planner need information about Nozzles ability to change Nozzles, automatically or not.
Single head Job planner have tree lists, problematic, unproblematic and sequenced.
Problematic need to be placed before sequenced. Unproblematic can be placed anytime,
and sequenced need to be placed in sequence.
After checking the used nozzles inside sequenced queue, it removes that parts from unproblematic
and problematic queues and prepend it to the sequenced queue if the same nozzle is used.
Now Job processor know, it need X head changes inside sequenced queue, for N parts and 
Y nozzleTips for the rest.
Planner then need to decide to use one NozzleTip  prefereable having at least N placements and
bundle parts from unproblematic queue to the sequenced list if machine have more then one nozzle.
This is for every additional nozzle the head have.
The nozzleTip choosen is based on distance difference from feeder position. the bundled component
is choosen in base of distance deviation from targed placement position to next feeder pickup.
Now there are M parts that need to paced using O Nozzles with at least Y nozzle Tips.
Sequencing now nozzleTip changing needs knowledge of what Nozzles are able to change it's
nozzleTip and how many double nozzleTips are available. 
It don't matter if automatic or manually and i sincery wounder why no manual nozzle
changing was written, or i missed that ?
Is there something, that i have misinterpreted on the api, or is the api intended to change ?

vonnieda, 2016-01-11:
@cri-s I think you can do everything you mention here with the current API. The Planner interface is expected not to know anything about how nozzle changes will happen. The idea is that if Nozzle.getNozzleTips() returns any nozzle tips then those are available for changing.

So, you should be able to get the information you need by checking getNozzleTips() on each Nozzle. You can check Nozzle.getNozzleTip() (singular) to determine which tip is currently in use.

I do think that this API needs a bit of rework to grow in the future, as mentioned in #183 but I think you can finish what you are doing with the current API and the changes in the future will be minimal.

As for manual nozzle tip changer, it's on the TODO list, just hasn't been done yet. It's #118.

cri-s, 2016-01-11:
Yes and no, for my point of view, for the planner is crucial to know
if nozzle can be changed
and to what nozzles. Otherwise it cannot plan in advance.
from the getNozzleTips() i know, and i dislike it, as it's misleading,
but can work with some
workaround. Further planner need have state machine and job processor
some interface to
it in order to know if planner can advance or not, or maybe job
processor advances state machine.

vonnieda, 2016-01-11:
@cri-s As I said, the Planner doesn't need to know *how* nozzle changes will happen, only that they *can* happen. It knows this by looking at getNozzleTips(). If getNozzleTips().size() > 0 then nozzle changes are available.

There is a small issue now where you might have multiple NozzleTips on a Nozzle and not have the changer enabled, but this is a combination of a bug and a configuration error. If you do not have a nozzle changer you should not have additional NozzleTips (configuration error) and if we have a manual nozzle changer then it's no problem (bug).

cri-s, 2016-01-11:
It is thinkable that nozzleTips have (internal) numbers, or can be
assumed that this are inside
id ? example 401 (400 serie nozzle, second digit is type, and third is number).

vonnieda, 2016-01-11:
Sure, that is possible. Either way is fine. If you need additional fields on the NozzleTip just extend ReferenceNozzleTip and add whatever you want.

closed 2016-01-11 by vonnieda