openpnp/openpnp #189: Discussion for Driver refactoring, moveTo blocking, GCode, etc.

vonnieda, 2016-01-11:
Comments originally made on #183, moved to this issue for further discussion.

 cri-s commented 2 hours ago
Jason, i have extended the driver api a bit,
void dwell()  , wait until finished
void dwell(double)  wait time in seconds.
boolean GCode(string)
boolean GCode(string,time)

Actually the only not right is on firepick driver, where i probably
need community help if this
get accepted by you. Then i suggest to use GCode for driving it. I
suggest then to use GCode
for driving changer and i think, manual should be highest priority.
Currently i have tested only dwell(double) in order to go away with
thread sleeping and that
works.

2016-01-11 17:12 GMT+01:00, Jason von Nieda <notifications@github.com>:
…
 @vonnieda
  Owner
vonnieda commented an hour ago
void dwell() , wait until finished
void dwell(double) wait time in seconds.

What is the purpose of the dwell commands and how will they be used? I don't see this as being included because all of the driver methods are supposed to block until they are finished. If something is returning before it's complete then that is a bug.

boolean GCode(string)
boolean GCode(string,time)

These we will not include. The Driver interface is designed to be controller agnostic and this would make it specific to GCode. If you want GCode specific methods just extend the ReferenceDriver interface and implement them in your branch.
 @cri-s
  
 cri-s commented 44 minutes ago
dwell wait until all gcode are processed. No only some gcode are
blocking, the other return without being completed.
dwell(double seconds) wait seconds until and return after all previous
motion code are finished.
I have used it to replace    Thread.sleep(pickDwellMilliseconds);  as
otherwise it don't worked
because gcode and pc are never in sync.
GCode() would be nice for nozzle changer code and others, that are
always machine specific
and mostly have only moves inside it. You'r decision.

2016-01-11 18:04 GMT+01:00, Jason von Nieda <notifications@github.com>:
…
 @vonnieda
  Owner
vonnieda commented 32 minutes ago
@cri-s All driver calls should block until they are finished. This contract must be enforced or lots of other things will break. For instance, if moveTo() or place() does not block then the machine may try to move before it's finished placing. If you have driver calls that don't block, that is a bug and should be fixed. Just call dwell() internally in the driver, as many of the drivers already do.

We can't add GCode specific things to the driver interface because there are machines that don't use GCode. For instance, the TinyG and FirePick drivers use JSON for communication. Having a GCode command on it would be meaningless.

If it's very important for you to be able to send generic commands to the driver I would be okay with adding something like public Object sendCustomCommand(Object);. With this you could do something like sendCustomCommand("M100") and then process that as Gcode within your driver.

That being said, I suggest you extend ReferenceDriver and add these functions to your own driver. The program is designed to be extended like this.
 @cri-s
  
 cri-s commented 11 minutes ago
moveTo don't need to block, and it's contra productive. It's enought
that after pick and place
the call is blocking, and as example actors are not blocking too. .
GCode works for TinyG , FirePick , don't know i have left it as dummy
function for now.

vonnieda, 2016-01-11:
@cri-s OpenPnP expects all driver interactions to complete before they return, i.e. block. Without this assumption there are many things that break in OpenPnP. Vision operations expect the camera not to be moving, pick and place operations expect movement to be complete, drag feed operations expect the solenoid to retract before moving, etc. This is a fundamental assumption in OpenPnP and it's not changing without a major redesign.

This is even in the interface documentation. See https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/machine/reference/ReferenceDriver.java#L32

As I said previously, if there is a function in the driver that does not block before returning currently then that is a bug that should be fixed.

closed 2016-01-13 by vonnieda

cri-s, 2016-01-13:
It is possible to use dwell(time) for all delays and on some utility class
the devell time
Is then converted to delay call.
This would allow coexistence with real gcode and wait until done as you
have actually made it. This development  was not noticed by me as the
default driver not works and the improvemnted was rejected on submission.

vonnieda, 2016-01-13:
@cri-s If you want this to be added please submit a clean pull request showing why it's needed and how it is used. From what you have described so far this seems to break the driver model as it is currently designed.

cri-s, 2016-01-13:
As pure discussion , here one Driver. The addition to ReferenceDriver.

	public  boolean GCode(String command) throws Exception ;
	public  boolean GCode(String command, long timeout) throws Exception;
	public  void dwell()  throws Exception;
	public  void dwell(double Seconds)  throws Exception;

And instead of calling Thread.sleep(Milliseconds) it calls
driver.dwell(Milliseconds/1000.)
whenever the thread need waiting.
As you see, only pick and place are blocking on Gcode level (waiting
for end of motion).
All other opereration are queued on execution.
Further ther exist gcodes, as example M121 that calls relay and wait
for external acnowledge
on F_IN input.
The point is, that like on cnc machine, the gcode is streamed down,
not send line by line and
waited for completition, otherwise it's really slow, and camera
capture cannot be made on
65ms, it need a lot more, if doing this on PC waiting.
Thats the whole question. Actually i call driver.dwell() , but dwell()
function, that call driver
itself should be more generally used. As said, i don't have actually
solution for firepick driver.
If you say yes, i ask, otherwise it's just dummy function.
pc timing on sleep can be worst, serail usb buffering implies delays, ... .

package org.openpnp.machine.reference.driver;

import java.io.IOException;
import java.util.ArrayList;
import java.util.List;
import java.util.Queue;
import java.util.concurrent.ConcurrentLinkedQueue;
import java.util.concurrent.TimeoutException;

import javax.swing.Action;

import org.openpnp.gui.support.PropertySheetWizardAdapter;
import org.openpnp.gui.support.Wizard;
import org.openpnp.machine.reference.ReferenceActuator;
import org.openpnp.machine.reference.ReferenceHead;
import org.openpnp.machine.reference.ReferenceHeadMountable;
import org.openpnp.machine.reference.ReferenceNozzle;
import org.openpnp.machine.reference.driver.wizards.AbstractSerialPortDriverConfigurationWizard;
import org.openpnp.model.LengthUnit;
import org.openpnp.model.Location;
import org.openpnp.model.Part;
import org.openpnp.spi.PropertySheetHolder;
import org.simpleframework.xml.Attribute;
import org.slf4j.Logger;
import org.slf4j.LoggerFactory;

public class MultiHeadMultiRelayDriver extends
AbstractSerialPortDriver implements Runnable{
	private static final Logger logger =
LoggerFactory.getLogger(MultiHeadMultiRelayDriver.class);
	private static final double minimumRequiredVersion = 0.81;
	private double connectedVersion;
	private boolean disconnectRequested;
	
	private Object commandLock = new Object();
	
	private double x, y, z, z0, z1, c, c0, c1, f=0;
	private boolean tool=false;
    @Attribute(required=false)
    private int mm=0;
    @Attribute(required=false)
    private double delay=0;
[... 494 lines cut ...]
                try {
                    line = readLine().trim();
                }
                catch (TimeoutException ex) {
                    continue;
                }
                catch (IOException e) {
                    logger.error("Read error", e);
                    return;
                }
                line = line.trim();
                logger.debug("<< " + line);
                responseQueue.offer(line);
    if (line.equals("ok") || line.startsWith("error: ")
        || line.startsWith("$VERSION =")) {
        // This is the end of processing for a command
//System.out.println("Line = " + line);
        synchronized (commandLock) {
            commandLock.notify();
        }
    }
}
}

}

vonnieda, 2016-01-13:
@cri-s Please submit a *clean* pull request that doesn't break other code and features and I will consider it. Otherwise just keep this change in your fork. See https://github.com/openpnp/openpnp/blob/develop/CONTRIBUTING.md