openpnp/openpnp #190: NozzleTip changer LocationButtons incorrectly moveToSafeZ, and use wrong tool.

vonnieda, 2016-01-12:
From this discussion: https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/5ND1yV4YFnM/1aZAzMuSCQAJ

Two issues, really:

1. With the recent refactor of tool selection, the LocationButtons are using the paste dispenser if that tab is selected. This doesn't make sense when working with a Nozzle/NozzleTip. We should pass a flag to restrict this on wizards where it's important.

2. In the same vein, using moveToSafeZ for NozzleTips changer is a mistake because it makes it impossible to manually sequence the three moves for testing. We can also pass a flag for this.

cri-s, 2016-01-12:
1) the tool selected on the drop down panel above coordinate displays should be used, do you speak about this ?
2) safeZ on beginning and end of nozzle changer sequence is fine.
   on the gui it should be button to go safe-Z , store position, recall position.

vonnieda, 2016-01-13:
#1 has been fixed ind037789a8c75c5e3bb0866e0fdc7e16e26039927. The location buttons will only select nozzles now, like they did previous to the recent change.

commit 3c98aee6 (2016-02-13): Add new "position nozzle without safe z" button option to location buttons panel and enable it for the nozzle tip wizard. This fixes #190.
Fixed copy and paste icons which were the wrong size.
Dumped the .idraw and .svg versions of the full icon sheet as these contained all the wrong sizes of icons. Better to just manage the icons individually.
closed 2016-02-13 by vonnieda