openpnp/openpnp #191: Fiducial check for board bottom fails by specifying a large angle.

vonnieda, 2016-01-14:
See discussion at https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/LPS0sW7ucKw/1SBMraJ_DAAJ
And video at https://www.youtube.com/watch?v=OkH-prmXbsY

I tested this using the pnp-test dataset with the image mirrored and got a similar result. The fiducials are located properly but then the angle gets set to 90ish when it should be zero. Suspect this is related to #148 and #186

cri-s, 2016-01-25:
can you retest it with latest fix_coord applied. Because of opencv bug, i cannot test it without modification of code

vonnieda, 2016-01-30:
Bottom fiducials were broken entirely until now. This should be fixed in 5276563869193d32d305256795b9fbc481667df7. I've tested with many different board setups and it's all working nicely. Please reopen if it doesn't solve the problem.

closed 2016-01-30 by vonnieda