openpnp/openpnp #193: Security error on new machine interface rewrite.

cri-s, 2016-01-20:
When STOP button on GUI is pressed, the driver MUST immediatly call setEnabled() on driver.
It cannot wait that home process or other queued move end, because the driver send Reset requests
to controller and this immediatly stops any motion or actor and report machine to safe state.
It cannot queued for future actions.

vonnieda, 2016-01-20:
Good catch, you are correct. This is a bug.

cri-s, 2016-01-20:
I have doing this, but as expected, test cases don't working anymore
and so i have deleted
test cases. How to fix it. Further clearing the Motion Queue don't work.
Annother point is with this motion queue. openpnp calc with
getLocation from nozzle, whatever. Now until move is not completed,
the calc is wrong. One example is enabling
driver , homing and then clicking to motion, as example nozzle, , or
anything that is relative
and use that for calc. Result: big crash.

   @SuppressWarnings("serial")
        private Action stopMachineAction = new AbstractAction("STOP") {
                @Override
                public void actionPerformed(ActionEvent arg0) {
                ReferenceMachine machine;
                        machine = (ReferenceMachine) configuration.getMachine();
                        setEnabled(false);
                        try {
                        machine.getDriver().setEnabled(false);
                        machine.setEnabled(false);
                        } catch(Exception e) {;}
                }
        };

commit f839aaa2 (2016-02-13): Fixes #193, setEnabled is now called in it's own thread instead of on the machine task queue. This keeps emergency stop situations from getting queued and putting it in it's own thread keeps it from blocking the UI.
closed 2016-02-13 by vonnieda