openpnp/openpnp #197: Board / BoardPlacement / SolderPad

cri-s, 2016-01-23:
It's possible to remove fiducial and solder pads as special thing and threat it just as Placements ?
solderPaste are just normal Placements, just annother Nozzle(Tip), and part and feeder can be SolderPaste with Location of the feeder to be Double.NaN .
Without pcb leveling the results are not good for fine pitch, using Standoff works.
Standoff allow only placing dots, not lines.
Fiducial are not well supported, it's better to remove them and use special Placements for that.

vonnieda, 2016-01-23:
No. As the solder paste feature grows the data will become more specific
and I don't want to clutter up Placement with it.
Right now fiducials are just placements, but that may change. I'm
considering making them their own thing so I can support them more
specifically.

cri-s, 2016-01-23:
Can you define the interface for Parts to fiducial and define the various fiducials? 
I'm tweaking the code to correct the fiducial/angle/... problem, or traying to implement it different.
Fiducial and futue Fiducial interface can play a role in this.

Inside Configuration , i have the follow function for delimiting the placement Area.
Important thing, no Feeder can (should) be configured inside that Area.

      public void setPlacingAreaMin(Location )
      public void setPlacingAreaMax(Location )

Should this function that are used by Gui and JobProcessor be located on the same Configuration,
or should it stay outside ?      
      public boolean isInsidePlacingArea(Location loc) {
        public Location map(Location loc) {
        public Location unmap(Location loc){

vonnieda, 2016-01-23:
Sorry, I can't define those interfaces right now. I haven't fully decided
how I want to do it.

Your placingArea stuff should go on a subclass of ReferenceMachine. I'd put
the utility functions there as well.

Jason

cri-s, 2016-01-23:
This placingArea stuff is used by Gui and JobPlanner .  Ok, if you say that, i place it there.

closed 2016-01-23 by vonnieda