openpnp/openpnp #205: Job Placements need a status field that shows the current status of that placement during job run.

vonnieda, 2016-02-02:
This would be a new field on the Placements table and it would need to be backed by data that combines BoardLocation and Placement. Statuses, to start, would be:
* Pending
* Working
* Complete
* Failed

I think this might just be a transient map rather than something we store. It would just get cleared when you start the job. In a later phase we may want to store the data so jobs can be resumed.

vonnieda, 2016-02-15:
@anthonywebb A question on resuming a job: Would you want/expect to be able do this between runs of OpenPnP? i.e. should the status be saved to the job file or something so you could exit OpenPnP, start it back up, reopen the job and pick up where you left off? If so, I assume we'd need to supply a way to reset / restart a job?

I guess one thing we could do is save the job status and then when you attempt to start an unfinished job prompt to either resume or reset it?

anthonywebb, 2016-02-15:
If I were to exit OpenPNP I would not expect to resume the job from where I left off.  I do like having a list of processes that I can jump to the middle of and continue from manually if needs be, but I would expect the job to start fresh when the job was first launched.

mikeselectricstuff, 2016-03-04:
You absolutely do need to be able to restart a job from where you left off. 
The system could have crashed or had a power failure.
You may have had a hardware/mechanical failure mid-job that needs a powerdown in order to fix it.
A feeder may have badly jammed so for speed you want to swap in another feeder to finish the job, which may require a powerdown or restart.
All info relating to the progress of a job sould be in the job (or another) file, and updated in a way which is reasonably robust ( e.g. the file would not be held open longer than needed)

vonnieda, 2017-09-08:
This feature is now mostly complete. The only thing remaining is saving the job on a regular basis.

vonnieda, 2018-07-31:
Job saving was added in a recent PR, so this is finally complete.

closed 2018-07-31 by vonnieda